FreeOpcUa/opcua-asyncio #297: Does opcua-asyncio servers support browse?

FabPi, 2020-09-24:
Hello everybody,
I am currently trying to connect a program to an opcua-asyncio server via OPCUA. Unfortunately, I always get an error while browsing. How do I add browse functionality to the server or is it not yet supported?
Thanks for your help!

[image]

swamper123, 2020-09-24:
Hello FabPi,

well browsing is supported and as you can see in other issues, there are even screenshots to prove it.
Have you checked the URL you are browsing to as Client and compared it to the server one?

FabPi, 2020-09-24:
Hello swamper123,

i just used the example server (simple-server) and tried to connect it to a program (Siemens NX). Programside i just copied the url from the server. I also tried to connect with Matlab instead and checked terminal output. Seems i made a mistake serverside? Even with `server.set_security_policy([ua.SecurityPolicyType.NoSecurity])` it denies the connection. Either Siemens NX or Matlab connection results in the following failure:

```
INFO:asyncua.server.binary_server_asyncio:New connection from ('127.0.0.1', 61706)
WARNING:asyncua.server.uaprocessor:Someone attempted to do something they are not permitted to do
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 61706)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('127.0.0.1', 61706), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('127.0.0.1', 61706)
```

swamper123, 2020-09-25:
Hmmm...sounds strange, because it's an BadUserAccessError. The server recognised a connection so the URL seems to be fine.
The Client (so your Siemens NX or Matlab) is trying to connect without security policy and user access as well?

AndreasHeine, 2020-09-25:
a wireshark trace would be helpfull!

FabPi, 2020-09-25:
Hello AndreasHeine,
i did a wireshark record on the local port. I dont exactly know, whats the focus of interest, so i recorded entire session ([wireshark_tracing_file](https://w-hs.sciebo.de/s/1CxmG8wRuGXHumz)).
In Matlab i do a simple opcua Connect, via:
```
uaClient=opcua('localhost',2030)
connect(uaClient)
```
Thanks for your help!

[image]

AndreasHeine, 2020-09-25:
ok in your trace are no opcua protokol messages means you use encryption!? what i am also missing is for example the opening handshake client hello -> server ack -> https://reference.opcfoundation.org/src/v104/Core/docs/Part6/readme_files/image016.png

[image]

swamper123, 2020-09-25:
If I interprete the [Matlab docs ](https://de.mathworks.com/help/opc/ug/opcua.html;jsessionid=4480f7b0000633f19d9fd153da3c) right, @AndreasHeine is right that an encryption is used.

You may have to set your UaClient properties the right way, since it seems that the ChannelSecurityPolicy is Aes256_Sha256_RsaPs as default.

FabPi, 2020-09-25:
What i've tried lately: Updated to Matlab 2020b and tried to set the SecurityPolicy in Matlab. This leads to an error just before there is a chance to set the security policy:
```
>> uaClient = opc.ua.Client('opc.tcp://localhost:2030/freeopcua/server/');
Error using opc.ua.Client/fetchEndpoints (line 684)
The operation timed out.

Error in opc.ua.Client (line 378)
                        [~, appName] = this(tI).fetchEndpoints;
```
Matlab Code Documentation for Security says:
```
%ChannelSecurityPolicy - The Channel Security Policy to be used for connections. Supported Channel
%Security Policies are defined in the enumeration class opc.ua.ChannelSecurityPolicies. The default
%value is 'Invalid'. When you construct an OPC UA Client with a valid hostname and port, or
%endpointUrl, the server will be queried for available endpoints, and the ChannelSecurityPolicy will
%be set to the highest security setting advertised by the server. Use setSecurityModel to change the
%selected MessageSecurityMode.
```

This Code throws the error message:
```
% Call the findServer service on the server. This returns the server's unique URL
            [serverList, errorCode ]= opc.internal.opcUaDiscovery('getServerList', discoveryUrl);
            if (getUaMessageSeverity(errorCode) == 2)
                % Check for specific error message, and translate if necessary
                msgId = getUaMessageId(errorCode);
                if (strcmp(msgId, 'opc:ua:StatusCodes:Bad_TcpInternalError'))
                    error(message('opc:ua:Client:GetServerListFailed', discoveryUrl));
                else
-->                 error(message(msgId));
                end
            end
```

This error appears with either the ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt, Basic256Sha256_Sign or NoSecurity asyncio server.

Any further Suggestions for testing?
The non async version works still fine with Matlab (something like the minimal server). I can produce the client, connect and retrieve Values as expected:
```
>> uaClient=opcua('opc.tcp://localhost:2030/server/')

uaClient = 

OPC UA Client:

   Server Information:
                     Name: 'FreeOpcUa Python Server'
                 Hostname: '127.0.0.1'
                     Port: 2030
              EndpointUrl: 'opc.tcp://127.0.0.1:2030/server/'

   Connection Information:
                  Timeout: 10
                   Status: 'Disconnected'
              ServerState: '<Not connected>'

   Security Information:
      MessageSecurityMode: None
    ChannelSecurityPolicy: None
                Endpoints: [1×1 opc.ua.EndpointDescription]

>> connect(uaClient)
>> serverNodes = browseNamespace(uaClient);
>> disp(serverNodes);
1×4 OPC UA Node array:
    index     Name      NsInd  Identifier  NodeType
    -----  -----------  -----  ----------  --------
      1    Temperature  2      2           Variable
      2    RFID Status  2      3           Variable
      3    RFID         2      4           Variable
      4    Time         2      5           Variable
>> [value, timestamp, quality] = readValue(uaClient, serverNodes);
```

AndreasHeine, 2020-09-28:
> The non async version works still fine with Matlab (something like the minimal server). I can produce the client, connect and retrieve Values as expected:

python-opcua works and asyncua not?

AndreasHeine, 2020-09-28:
actually the only difference on asyncua is that asyncua has some changes from @JoeyFaulkner certificate stuff, would be importent do look closer on that!

FabPi, 2020-09-28:
> python-opcua works and asyncua not?

I tried the python-opcua a while ago and made a minimal server with some custom nodes and variables for sensor-values. This one works fine with Matlab. The minimal asyncua server does neither connect to Matlab nor Siemens NX (Mechatronic Concept Designer) with the standard procedure i used for the python-opcua server.

JoeyFaulkner, 2020-09-28:
hmm this is interesting, I am really busy today but will try and find time to look at this tomorrow. It doesn't make any sense that the NoSecurity server would throw this error. I know we had a issue similar to this a little while ago, but it was fixed: are you up to date with the current master here?

FabPi, 2020-09-28:
> hmm this is interesting, I am really busy today but will try and find time to look at this tomorrow. It doesn't make any sense that the NoSecurity server would throw this error. I know we had a [...]

I've  checked the version of my virtual enviroment with pip freeze: asyncua==0.8.4 and opcua==0.98.12. So it seems up to date.

JoeyFaulkner, 2020-09-28:
did you install using `pip install asyncua`? could you try uninstalling that and reinstalling with `pip install git+https://github.com/FreeOpcUa/opcua-asyncio`? :)

FabPi, 2020-09-28:
> did you install using `pip install asyncua`? could you try uninstalling that and reinstalling with `pip install git+https://github.com/FreeOpcUa/opcua-asyncio`? :)

This did the trick. Now everything works. I am able to connect, browse namespace, etc. with Siemens NX and Matlab.
Thanks for your help!

JoeyFaulkner, 2020-09-28:
amazing! glad it works :) @oroulet it might make sense for us to iterate the version number

closed 2020-09-28 by JoeyFaulkner

oroulet, 2020-09-28:
@JoeyFaulkner what is the issue? I did not follow that thread

JoeyFaulkner, 2020-09-28:
it was an issue fixed in https://github.com/FreeOpcUa/opcua-asyncio/commit/46117c00b142a35bc4289e1ac726ec1658ef2440 but it is the same version number as the current one and was installed via pip :) makes it a little harder to debug.

oroulet, 2020-09-28:
I am new release 0.9.0

JoeyFaulkner, 2020-09-28:
awesome, thanks! :)