FreeOpcUa/opcua-asyncio #298: ModifySubscriptionRequest_Encoding_DefaultBinary missing

joncutting, 2020-09-24:
Hi,

The uaprocessor in opc-asyncio does not support ModifySubscriptionRequest_Encoding_DefaultBinary like its python-opcua cousin.   I'm not sure why that is but would recommend to implement it. 

As far as I can tell this requires minor modifications to uaprocessor.py, internal_session.py, and subscription_service.py.

Jonathan

AndreasHeine, 2020-09-24:
that could be because python-opcua is in maintenance mode and some one made a PR only for python-opcua.
typically asyncua is the latest and most featured version because of its performence and readability!
tbh i have never seen a OPC UA Server which uses a different encoding then binary-protocol (most efficient)...
websocket-protokol is not implemented yet in which xml and json encoding would be best.

joncutting, 2020-11-11:
Just submitted PR#330 earlier this evening. This is a real issue that is preventing client software based on a commercial OPCUA library from connecting to an asyncua server (it works fine with non-asyncua server). Note that this is not changing the protocol to XML, it remains binary.