FreeOpcUa/opcua-asyncio #299: Make lxml dependency optional

joncutting, 2020-09-24:
Hi,

As far as I can tell, the package lxml is only used for XML import/export, but it is still a requirement even if that feature is not needed. Normally I would say no big deal but for frozen code (e.g. pyinstaller) it adds an extra 2MB to the executable which means additional delay during startup. If there is any way to make lxml optional - or only imported when it is really needed - then it would be nice to have.

Jonathan

AndreasHeine, 2020-09-24:
thats difficult... not the removal but if you use OPC UA in the way it is designed for, means to implement standardised interfaces/information models via OPC UA Companion Specs. (pure XML-Nodesets) dont get me wrong feel free to PR but most people use Companion Specs. for modeling so its abolutly necessary and a core part of OPC UA!

joncutting, 2020-09-24:
OK, understood. There are many opinions out there about how best to lazy load, so I took the lazy way and grabbed `[lazy_import](https://pypi.org/project/lazy-import/)`. I submitted PR #300.

swamper123, 2020-09-25:
I am not very happy with lxml as well, because there may be some security issues in this library (I may need to investigate further into that topic) about xml attacks. But this library is still not in V1.0 so this will be a topic in the future.
If you want to load the addressspace in a fast way, you may have a look at:
https://github.com/FreeOpcUa/opcua-asyncio/blob/582f3a2065b08cce5e4daa5d7b8272e92e1856d0/asyncua/server/address_space.py#L590
Where you can load a pickled Server into the address space. This is fast, but in my opinion not better than the lxml way (mentioned as well in #277 )

swamper123, 2020-12-07:
With #346 this dependency is now removed. Case closed 🕵️

closed 2020-12-07 by swamper123