FreeOpcUa/opcua-asyncio #301: FindServersOnNetwork -> BadUserAccessDenied

reida, 2020-09-25:
**Steps to reprouce.**
1. Launch UA-Expert (v.1.5.1)
2. Launch examples/server-minimal.py
3. In UA-Expert add new server via Custom Discovery Service

**Ua Expert log:**

`
Discovery FindServersOnNetwork on opc.tcp://localhost:4840/ failed (BadUserAccessDenied), falling back to FindServers
DiscoveryWidget|Adding Server FreeOpcUa Python Server with URL opc.tcp://0.0.0.0:4840/freeopcua/server/
`

**Console Log:**

`
INFO:asyncua.server.binary_server_asyncio:New connection from ('192.168.0.3', 63999)
WARNING:asyncua.server.uaprocessor:Received a request of type 12208 without an existing session
WARNING:asyncua.server.uaprocessor:Someone attempted to do something they are not permitted to do
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('192.168.0.3', 63999)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('192.168.0.3', 63999), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('192.168.0.3', 63999)
INFO:asyncua.server.binary_server_asyncio:New connection from ('192.168.0.3', 64002)
INFO:asyncua.server.uaprocessor:find servers request (None)
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('192.168.0.3', 64002)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('192.168.0.3', 64002), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('192.168.0.3', 64002)`

imho the problem in nodeID 12208. if i do the same with opcua library, i will get 
`Discovery FindServersOnNetwork on opc.tcp://localhot:4840/ failed (BadServiceUnsupported), falling back to FindServers`
`Unknown message received FourByteNodeId(i=12208`

I need to add new sever via Discovery Service becouse my client is using Siemens OPC Scout ( i have not access to this software for debug opcua communication), and he told me, that he can add new server endpoints ONLY with discovery dervice.

I have tried to add my server to the Local Discovery service (https://opcfoundation.org/developer-tools/samples-and-tools-unified-architecture/local-discovery-server-lds/), but no success. Many errors. But it is for another issue, that already exists

AndreasHeine, 2020-09-25:
https://opcuastack.readthedocs.io/en/latest/3_development/discovery_process.html#local-discovery-server

your OPC UA Server needs to register itself to the LDS Server!

AndreasHeine, 2020-09-25:
https://github.com/FreeOpcUa/opcua-asyncio/blob/9f3dd531397db854f2e1d2a59ff04f4231cab418/asyncua/server/server.py#L206

reida, 2020-09-25:
Thank you for the quick answer. I will try

reida, 2020-09-25:
> https://github.com/FreeOpcUa/opcua-asyncio/blob/9f3dd531397db854f2e1d2a59ff04f4231cab418/asyncua/server/server.py#L206

It is solution. Problem resolved. Very thanks!

closed 2020-09-25 by reida