FreeOpcUa/opcua-asyncio #304: Problem to get Alarms Event from SIEMENS 840D sl SINUMERIK OPC UA Server

cepinto, 2020-09-29:
Dear All,

I have no data from the event alarms subscription in the OPC UA server from Sinumerik 840D sl , I try from diferents examples and its not working, no events alarms was returning from the Server .... When a use the UA Expert in the same node that Im trying to subscription the events , it works... 
Someone have any idea what's happening ?

The code is bellow :

# handler is the reference os my class that has the function def event_notification(self, event):
frequence = 1000
nodeId = "ns=2;i=1" #Sinumerick root Nodeid
self.event = await self.client.create_subscription(frequence, handler)
node = self.getNode(hwId)
await self.event.subscribe_data_change([node])
myevent = await self.client.nodes.root.get_child(["0:Types", "0:EventTypes", "0:BaseEventType"])
self.evhandle = await self.event.subscribe_events(node,myevent)

For the subscription data change its work fine with the Server …

Follow attached the print screen from UA Expert of event alarms View 

[image]

Looking for the examples , the events are subscription with sucess , but no data are came from Sinumerik OPC UA.

AndreasHeine, 2020-09-29:
does the node you subscribe ("node") has a reference "GeneratesEvent" ??? screenshot if possible!

cepinto, 2020-09-30:
Hi , sorry only tomorow I will be in the machine and I will get pictures Using UAExpert ,  but the Sinumerik "node" act as a root node for all events alarms, and it has the "EventNotifier" attribute. According with the Siemens documentation :

"Events are by themselves not visible as nodes in the address space. They can only be received via objects. Not all objects can signal events. Whether an object can signal events is specified at the object by the EventNotifier attribute. Only objects where this attribute has been set can be specified in the Event Monitored Item and received in Clients Events.

AndreasHeine, 2020-09-30:
ok! i have to ask some questions ;) to get a better view.

Some events will always be fired from server object only but i dont know much about simens sinumeric (just siemens plcs)!
"0:BaseEventType" is it realy a base event or a siemens specific?
Alarms & Conditions is not fully implementet at this point but we support events so it should work!
I guess you have looked over the examples!

waiting for the screenshots :)

cepinto, 2020-10-01:
Hi @AndreasHeine !

Today the CNC Machine its under maintenace and the OPC UA Server its not running , I will try to take pictures tomorow ....

Aswering the question , I read the documents and the alarms are of type “CNCAlarmType” that is derived from "ConditionType" but I believe that the root of the derivation hierarchy is the BaseEventType.

Sorry i'm learning regarding OPC UA ....

According to SIEMENS document:

"The SINUMERIK Alarm object is of the “CNCAlarmType” which is defined in the Companion
Specification “OPC UA Information Model for CNC Systems (http://opcfoundation.org/UA/
CNC/)”.

The root of the derivation hierarchy is the BaseEventType. The types for Alarms and Conditions
are available below the ConditionType. The Application-specific event types such as
CncAlarmType can be derived.

The basic type for all condition objects is the condition type. It is derived from BaseEventType.
All mechanisms for alarm processing work even without the condition objects are contained
in the address space. "

AndreasHeine, 2020-10-01:
all events dervie from "BaseEventType" but you need to listen to a specifc eventtype ("CNCAlarmType")

```python
myevent = await self.client.nodes.root.get_child(["0:Types", "0:EventTypes", "0:BaseEventType","CNCAlarmType"])
```

you will only recv those events of the type you "subscribe_events" to!

oroulet, 2020-10-01:
If subscribing to BaseTypeEvent you should receive everything.. if not there is something strange..

AndreasHeine, 2020-10-01:
`If subscribing to BaseTypeEvent you should receive everything.. if not there is something strange..`

only if evfilter=None!?