FreeOpcUa/opcua-asyncio #308: import_xml fails on 0.9.0 (worked on 0.8.2)

jcbastosportela, 2020-10-07:
Great job, first of all, nice to see this project so active!

I have been using asyncua for a while now, but recently I noticed that the ServerTimestamps where being returned as "0", so I decided to try out the most recent version on asyncua via pip (0.9.0).

After the update, on the top of a series of **"warnings"** about `Message: 'overwriting parent target, shouldbe fixed'` that I did not yet investigate there is an **error**:
`asyncua.ua.uaerrors._base.UaError: Non array Variant of type VariantType.NodeId cannot have value None`

This error appears related with one of my custom VariableType. This VariableType's relevant settings are:

[image]

I have already attempted to change the **ValueRank** from _Any_ to _Scalar_ (as on  the attached picture), but problem persists.

Apart from the existence of the child properties this variable type is similar to the **ServerVendorCapabilityType** from the namespace 0 (http://opcfoundation.org/UA) (NumericNodeID 2137). I also attached that.

[image]

As you may have noticed I am using UaModeler from Unified Automation for developing my information model.

**The question:** Is this new behavior expected?

Note: I am using asyncua for my simulator of an OPC UA server that in reality is a PLC. The information model in the PLC works fine, as it does on 0.8.2

AndreasHeine, 2020-10-07:
https://github.com/FreeOpcUa/opcua-asyncio/issues/307

jcbastosportela, 2020-10-07:
Sorry all, just realized that my issue is also being mentioned in #307 
I have just read the title and first comments, but after sawmper123 reports the same issue.
Anything I can do for helping sorting this out?

swamper123, 2021-02-12:
@jcbastosportela Could you test, if this issue still exist with the current master branch? In case the problem is gone, we could close that case. :detective:

swamper123, 2021-04-21:
Since no complains happenes from @jcbastosportela : Case closed - 🕵️

closed 2021-04-21 by swamper123