FreeOpcUa/opcua-asyncio #311: Connect Ethernet Device to OPC Server (Question)

nunchuckBoP, 2020-10-12:
Hello, I am looking for some documentation / example on how to connect an ethernet controller that supports OPC-UA? I am hoping that this is still an active project?

Anyway, I need to connect a controller on ethernet to the server. I am assuming that when it gets connected all of the folders are automatically populated?

Thanks,
CJ

swamper123, 2020-10-12:
Hey nunchuckBoP,

well if you browse the server as a Client and you got the matching rights for that, you can request single nodes and it's attributes or multiple nodes with their attributes you wanted.
There shouldn't be a feature which is delivering all nodes of a server, because this costs time and other ressources.

Or did you mean with folders something else?

Perhaps you may have a look in our [Client-examples](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-minimal.py) (but you may have to replace the target Node with a Node of your server, the IP as well).

ccscpanici, 2020-10-12:
The examples are good and they show very well how you connect clients (Computers) to the OPC server and get simulated data in and out of it. I need to go one step further though and connect a device (PLC controller on ethernet) to the OPC Server that way I can get data in and out of the controller through the OPC server.

I am happy to write some driver wrappers for this for the controllers that I require (A-B ControlLogix, CompactLogix). I thought a good starting point for that would be to connect the device as a generic OPC device and see what data comes out of it.

Thanks,
CJ

ccscpanici, 2020-10-12:
Just as an example, this is how Inductive Automation connects these controllers to their OPC server.
https://docs.inductiveautomation.com/display/DOC79/Connecting+to+ControlLogix+v21

swamper123, 2020-10-12:
Well in such a case, you need to implement a Client in your controller software in some kind of way.
In the link, the company is using a Gateway-Layer, which seems to handle the PLC-Server stuff and handling the Gateway program. But in the end a Client is implemented there and has to request the Server anyway...like all Clients have to do, if they want informations from a server.

ccscpanici, 2020-10-12:
Not possible to inject software inside of the controller. Or are you saying setup my PLC controller as a client software piece connecting to the server? This is where I get confused on the practical applications of this library.

swamper123, 2020-10-12:
I am confused a little bit, so I will try to describe it in another way:

I expect such a setup, as you described:

|Program<OPC UA Client>| ---{NETWORK} -- {|your controller|--|target device <OPC UA Server>|}

You got a Device with an OPC UA Server. This server has an url you can browse and usually shall use Port 4840. This is working similar like other socket stuff as well (IP, Port).

Now you got another Program, which shall do stuff and talk with the Device. So you need an OPC UA Client, which connects to the OPC UA Server. Then you can get access to all the Nodes you want (usually).

So I guess the ethernet controller may just do it's ethernet controller thing?!

ccscpanici, 2020-10-12:
Attached is a picture of what I am exactly trying to do. Now, what you said is interesting...I think you were getting towards connecting to the controller directly with the client software. When I do that, however, on the port that I see in the Inductive Automation example I get a Timeout. 
[image]
[image]

AndreasHeine, 2020-10-12:
i guesssomeone did this mirror stuff already you should check this out: https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-ua-python-mirror.py

ccscpanici, 2020-10-12:
I don't see how that connects a PLC controller with the OPCUA Server software...

AndreasHeine, 2020-10-12:
your plc is a opcua server!? your opcua server can have a client part inside which reads and subscribe (mirrors) objects of the "plc opcua server" and your "data app opcua client" thinks your "opcua server" is the plc or at least the important parts thats a quit common usecase because plc tend to allow not much connections (less ressources) and often you have a data gathering layer between the plc and a large network from which maybe hundreds of clients connect and read data!

ccscpanici, 2020-10-12:
I want to connect an Allen Bradley ControlLogix PLC and read and write data in and out of it.

AndreasHeine, 2020-10-12:
which version of asyncua did you use? this error seems familiar!

we had several times problems with the max messagesize... there is a opening handshake in opcua on which the server and client negotiate the buffersize, messagesize... i work a lot with siemens s7-1500 opcua servers so maybe we need to investigate a little more with wireshark but lets first try connecting with uaexpert!? does it work with uaexpert?

ccscpanici, 2020-10-12:
It will not connect using UAExpert BadTimeout. It could be that I have the wrong configuration. The only really "documentation" that I have is the IP address and the slot number.

AndreasHeine, 2020-10-12:
then its very likely that the plc ignores the opening handshake... the only thing you can do is change the buffersize messagesize and chunksize manually in ua expert till you can connect or you can ask allen bradley for the right values but it should work automatikally like with siemens plcs...

ccscpanici, 2020-10-12:
Or could it be that the traffic is encrypted and A-B only gives out the key to software companies that pay millions of dollars for it? I have a working OPC server (RSLinx). I will wireshark it.

AndreasHeine, 2020-10-12:
but opcua is enabled and configured on plc side?

ccscpanici, 2020-10-12:
Yeah, opc-ua is always enabled on the PLC side. You can't configure OPC-UA on the PLC side, it's already there. Not sure what the configuration is though...that's the trouble.

AndreasHeine, 2020-10-12:
the only thing i found out that they are using the softing opcua stack, i havent testet python opcua client against the softing stack... unluckily i have only access to a softing opcua client (embedded in emulate3D) to test things...

ccscpanici, 2020-10-12:
Well, this is interesting. I decided to use my local Ignition OPC server to connect to the PLC. They have a "CIP Connection Size". Now, CIP is different than OPC-UA, could they be serializing CIP to OPC-UA?
[image]

AndreasHeine, 2020-10-12:
as far as i know they develope a companion spec for cip objects but all plc vendors i know use the binary protokoll which we support!

ccscpanici, 2020-10-12:
Well I was able to wireshark it - but I am not sure what I am looking at. Looks like a bunch of gibberish.

[image]

AndreasHeine, 2020-10-12:
encryption? if not that does not look like opcua binary...

ccscpanici, 2020-10-12:
Yeah, I can tell this is going to be a nightmare. However, there is another way...maybe the pylogix python library would be a better fit? It goes right to the controller through CIP. Not sure if it is as efficient as OPC, but something tells me there is a CIP layer to the Allen-Bradley OPC protocol...

AndreasHeine, 2020-10-12:
why does allenbradley does this... opcua is about interoperablility.... atm 99.9% of all plc vendors use opcua binary over tcp... silly in my opinion!!!

ccscpanici, 2020-10-12:
I totally agree with you. They make things way too hard in order to earn more $. Luckily there's a guy already out there that cracked the CIP code. :-)

AndreasHeine, 2020-10-12:
hey allenbrafley the O in opcua stands for OPEN 🤣

swamper123, 2020-10-13:
> why does allenbradley does this... opcua is about interoperablility.... atm 99.9% of all plc vendors use opcua binary over tcp... silly in my opinion!!!

If they are realy not using/supporting binary, it wouldn't be even OPC UA conform afaik 🤔  
TCP Binary is mandatory for every server afaik.

closed 2020-12-09 by oroulet

zsamora, 2022-10-26:
you managed to solve it? can you share code or at least explain which examples you used to develop the solution?