FreeOpcUa/opcua-asyncio #314: [Topic] Support for HA Client

ZuZuD, 2020-10-14:
I wonder if the community has already considered integrating a high-availability client like described in the `Part 4 section 6.6.2.4.5` of the protocol specification to this library.

For quite some times now, I worked with a HA Client based on the python-opcua client. I am currently rewriting this on top of asyncua and I aim to support the ```Hot``` mode (```6.6.2.4.5.4```) and from there, the ```Warm``` (```6.6.2.4.5.3```) mode shouldn't be too much of a lift.

In a nutshell, the spec describes a higher level client which establishes connections to two identical servers, and promote the server with the highest healthy service level (> 200) as the primary. The difference between primary and secondary being the publishing mode (enabled/disabled) for the subscriptions. Also, the sampling must be activated on the secondary for `Hot`, but disabled for the`Warm` mode.

I plan to have an abstraction layer with an API similar to the Client. A user could then leverage it to create an ideal configuration that will be subsequently applied to the lower clients according to their role. We could use several tasks to:

1. Check the service level
2. Promote a primary and reconnect clients individually 
3. Regularly merge the ideal and real configurations when needed.

I would like to know if there's people interested with such a feature and if this could be a good fit within the project. If that's the case, I could suggest some PR to discuss the details.

swamper123, 2020-10-15:
Since we are striving to a full OPC UA implementation, we would welcome every PR that brings us a step further.

This kind of Client/Behaviour is new for me so I may need some time to read and (hopefully) understand the specs (https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.2/#6.6.2.4).

ZuZuD, 2020-10-15:
> Since we are striving to a full OPC UA implementation, we would welcome every PR that brings us a step further.
> 
> This kind of Client/Behaviour is new for me so I may need some time to read and (hopefully) understand the specs (https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.2/#6.6.2.4).
> 
> EDIT: I am not even sure if we may have a handling for ServiceLevels in the Server?!

I used a different server so far, however from the latest build it seems we do not support it:
```
In [1]: from asyncua import Client, Node, ua
    ...: c = Client(url="opc.tcp://0.0.0.0:4840")
    ...: await c.connect()
    ...: assert await c.get_node(ua.ObjectIds.Server_ServiceLevel).get_value() is None
```

swamper123, 2020-10-15:
So this would be another point on our Agenda here as well.
Since I haven't got any other kind of server with that feature, at the moment, I may not be a great help in that case, but I guess some others here are being able to test the Client. And if that stuff is running, the server can be the next target.

oroulet, 2020-10-15:
: assert await c.get_node(ua.ObjectIds.Server_ServiceLevel).get_value() is None
We do not fill most of these values so this is not very surprising. Also that one has to be filled by every ua provider depending on hardware, network etc... we do not need to fill up some defualt value anyway I suppose

Also concerning the high level client spec: I have not read that part of spec at all but its sounds a bit strange. Usually high availability is attained using load balancing on server side. Doing that on client side sounds like a very strange solution... I woud personally really not prioritize that kind of things. There are many other more interesting part of the spec to implement (like the "real-time" part using UDP) but if you write good code and do not make the existing client more complicated I will merge it anyway.

AndreasHeine, 2020-10-15:
> : assert await c.get_node(ua.ObjectIds.Server_ServiceLevel).get_value() is None
> We do not fill most of these values so this is not very surprising. Also that one has to be filled by every ua provider depending on hardware, network etc... we do not need to fill up some defualt value anyway I suppose

if you use siemens plc as opcua client and you want to connect to python opcua server you need to set the servicelevel value... a default "255" would be better because the siemens checks cyclic serverstatus and servicelevel! other stacks initialise it as well (just loud thinking) could be a good point to add them?

swamper123, 2020-10-16:
A default 255, with a logger warning and a #TODO notice for the start. Then this Node would be useable and may not causing unknown behaviour (I mean like the example from Andreas kind of behaviour, which may not be explainable for us, not the "the server is drinking my beer, deleting my "Homework-folder" and stealing my car"-behaviour).

An idea for later would be a simple algorithm (because it's server specific) which is changing serviceLevel every ... minute & in case error happens?
Perhaps something like:

For 200-255 it can be a scale of connected clients.
2-199 if timeouts/errors happenes inc. client amount.
1 - Server is running, but not operational (perhaps if an amount of  xyz errors happen and the servers shall stop and perhaps go into maintainance mode)
0 - Maintainance (perhaps to make some cleanups/updates in the background or after a stop to get rid of node requests and start again? or a server admin is running updates/upgrades kind of thing)

oroulet, 2020-10-16:
no problem. PAtch welcome to fix these small things

AndreasHeine, 2020-11-11:
has anyone stept over 6.6.3 Client Redundancy?
https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.3/

> Client Redundancy is supported in OPC UA by the TransferSubscriptions Service and by exposing Client information in the Server diagnostic information. Since Subscription lifetime is not tied to the Session in which it was created, backup Clients may use standard diagnostic information available to monitor the active Client’s Session with the Server. Upon detection of an active Client failure, a backup Client would then instruct the Server to transfer the Subscriptions to its own session. If the Subscription is crafted carefully, with sufficient resources to buffer data during the change-over, data loss from a Client Failover can be prevented.
> 
> OPC UA does not provide a standardized mechanism for conveying the SessionId and SubscriptionIds from the active Client to the backup Clients, but as long as the backup Clients know the Client name of the active Client, this information is readily available using the SessionDiagnostics and SubscriptionDiagnostics portions of the ServerDiagnostics data. This information is available for authorized users and for the user active on the Session. TransferSubscriptions requires the same user on all redundant Clients to succeed.

ZuZuD, 2020-11-18:
> has anyone stept over 6.6.3 Client Redundancy?
> https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.3/

This is a neat feature that I would like to see working. Although, sharing the user authentication with a different client (backup) seems tricky security-wise.

closed 2023-03-29 by schroeder-