FreeOpcUa/opcua-asyncio #316: XML protocol not implemented yet?

jnsw, 2020-10-19:
The README.md states, that neither the client nor the server has the XML protocol implemented yet. But the example "simple-client-server-xml" uses an XML-file for the NodeSet. Is this just old information or is there something else meant by "XML protocol"? Or maybe I missed something. 

Thanks in advance!

AndreasHeine, 2020-10-19:
xml import und export of nodesets is supported!

xml and json protocol not!

AndreasHeine, 2020-10-19:
Overview about TransportProtocols:
https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.1/

TCP (Binary) is avalible!
https://reference.opcfoundation.org/v104/Core/docs/Part6/7.2/

SOAP/HTTP (Note: Deprecated in Version 1.03 because WS-SecureConversation has not been widely adopted by industry.)
https://reference.opcfoundation.org/v104/Core/docs/Part6/7.3/

Websocket and UDP based transport is also not avalible!

jnsw, 2020-10-19:
Thanks a lot! I am closing now.

closed 2020-10-19 by jnsw