FreeOpcUa/opcua-asyncio #322: Tests share same Server per testfile

swamper123, 2020-10-28:
Hello together,

I mentioned in #306 that I got some problems with the tests, which interfered eachother.

While testing the stuff through, I thought just changing the fixture to "function" would solve the problem.
But I suspect, that there is a memory leak happening which causes Travis freeze to death (10min. Timeout without response).

I would recommened that either each test cleans up behind itself and finishes clearly (not causing "Address already in use" and similar errors) or for every test a fresh server has to be buildUp and tearedDown afterwards somehow (which I would prefere and accept longer test runs).

swamper123, 2020-11-23:
*ping*
Any comments on this?
Shall we rewrite the existing tests?
Should we make better usecases where tests are allowed to influence each other in strict areas?
Does it make sense?
Pro/Cons?

ratara, 2021-04-25:
One of the Golden Blues of unit testing is that tests have to be independent of each other. Breaking this rule breaks the test in so many ways, it should always be a red line.

I highly recommend that the test are setup and teared down correctly including the server.

oroulet, 2021-04-25:
@ratara thank you for you impressive insight. Now i am waiting for you PR

oroulet, 2021-04-25:
@swamper123 yes many tests share the same server (not all) due to the performance reasons.
And yes some cleanup and more machinery to clean up after each tests would be nice. There are a bunch of functions already but i do not have overview anymore

ratara, 2021-04-25:
@oroulet I am no native speaker but i got it. :-)

Well i am not sure how much work is to do but i started a try. I got the code and now I am trying to understand whats going on.
1. It seems as there are missing dependencies like dateutil and more which are not mentioned in the dev_requirements.txt. Btw. there are no fixed versions of the dependencies, we should fix that to get no surprises.
2. I invested only 10 minutes to look but i did not found any howto setup the development system. This would help a lot.
3. I am new to pytest but after running a test discovery over 600+ files have changes. Is that some sort of bug or wrong usage by me? - I used VScode and the integrated commands for testing.
4. I feel like flooding the thread. Is there any place to talk about details without disturbing this thread?

oroulet, 2021-04-25:
1- this software is written by me and others when we have time, so yes many things are missing. just send PR for every things you find
2- no doc but you can look at setup.py file. It list all dependencies that are known
3. writting here is fine, or you can start a discussion: https://github.com/FreeOpcUa/opcua-asyncio/discussions

ratara, 2021-04-25:
Its working now, tests are running. I think for today that's it. I will start a PR for a complete dev_requirements.txt soon. I have to setup my git configuration first.
I really like your work and use the sync opcua project in my current project at work. It is a good peace of work and easy to use. Thanks for the work of you and your colleagues!

swamper123, 2021-04-26:
@oroulet Can coverage show up untested methods/functions (so comparing the project with the text file)? Such a thing would help us for the already existing and future ones.
I once tried to strip the tests like @ratara mentioned and it took centuries. At the moment we got 552 tests. A server needs to be raisedUp and tearDowned (on my device) ~ 6-7 seconds (let's say 6.5). This times 522 is 3588 seconds or around nearly an hour, just to start/stop a server...and then it's a typo in the end.
I haven't used the pickle loading way before. Perhaps a "start one time normally and then use the pickle file" would be faster? Haven't tried this before, because I am not a fan of pickles (except on my burger or potato salad).

oroulet, 2021-04-26:
not exactly sure what you are asking for. doyou know `py.test --cov=asyncua --cov-report=html` or `pytest -v -s -k my_test_name` ?

schroeder-, 2023-03-29:
Is fixed in current test setup

closed 2023-03-29 by schroeder-