FreeOpcUa/opcua-asyncio #334: Redo skipped tests

swamper123, 2020-11-18:
There are 4 Tests being skipped of different reasons:

Since the XML file is given from [UA-Nodeset master branch](https://github.com/OPCFoundation/UA-Nodeset), we can't test the xml file itself if a Node is missing or similar. This is a trust thing we have to accept, since we can't change them. We just can report [issues here](https://apps.opcfoundation.org/mantis/my_view_page.php) and yes there are issues existing in these files.

I guess this test shall show, that the Number of XML Elements containing Aliases in a file is the same as Reference Nodes being created out of the same XML file...not sure if this is usable. 
- https://github.com/FreeOpcUa/opcua-asyncio/blob/9925e7d098f8949d3f928ac8fb72fe469c4c07c1/tests/test_standard_address_space.py#L54

I dont't know what with "theat" is meant. 😕 
- https://github.com/FreeOpcUa/opcua-asyncio/blob/9925e7d098f8949d3f928ac8fb72fe469c4c07c1/tests/test_custom_structures.py#L90
- https://github.com/FreeOpcUa/opcua-asyncio/blob/9925e7d098f8949d3f928ac8fb72fe469c4c07c1/tests/test_custom_structures.py#L321

This test pass. Perhaps this was fixed with #130 
- https://github.com/FreeOpcUa/opcua-asyncio/blob/9925e7d098f8949d3f928ac8fb72fe469c4c07c1/tests/test_xml.py#L67

oroulet, 2020-11-18:
@pytest.mark.skip("support for theat currently removed")   => Support for that feature currently removed

that was probably a typo but I remember writting it

swamper123, 2020-11-18:
😄  okay, that would make sense.