FreeOpcUa/opcua-asyncio #337: Bug: cannot stop client if no connection established

joristork, 2020-11-18:
Steps to reproduce:
1. Ensure that the client is *not* able to connect to the given opc-ua server URI
2. `client.connect()`
3. `client.disconnect()`

The client will raise an `AttributeError`.
If one catches this, the asyncua thread hangs.

swamper123, 2020-11-18:
So the first connect request isn't successfull or is still running (from your step1)?
And then you are start another connect request?

Can you provide a small example code (eventually commented)? That would clear some questions and missunderstandings.

JuBanaSh, 2021-01-20:
Hi,
I'm facing an similar issue. I would like to retry the client.connect indefinitely as long as the connection is not established but it's not working. Here's my code. What am I doing wrong?

#Import des bibliothèques nécessaires à l'exécution
import os, datetime, time
from goto import goto, label
from opcua import Client

#Boucle infinie
while True:

        label .début
        try:
                #Configuration de l'URL d'accès au server
                url = 'opc.tcp://172.25.212.10:4840'
                #Etablissement de la connexion
                client = Client(url)
                client.connect()

        except ConnectionError:
                print('en attente de connection')
                time.sleep(5)
                goto .début
                
        else:
                print('Client connecté')

AndreasHeine, 2021-01-20:
if you have questions use the discussions feature of this repo (your post is not related to the topic...) if you open a new discussion please provide consoleoutput and stacktrace, just posting your code is not enough

JuBanaSh, 2021-01-20:
@AndreasHeine Sorry, I will correct it.

dan-sel, 2021-02-18:
@joristork The asyncua thread doesn't hang, it was never told to stop. The exception is raised in the `self.tloop.post(self.aio_obj.disconnect())` call in `asyncua.sync.Client.disconnect`. This means that the code that stops the thread loop is never called. Quick fix: manage the thread loop yourself.

Randelung, 2021-04-06:
Pretty sure it's just a case of 'forgot to check if the task exists that I'm trying to cancel'. My PR allowed me to gracefully close the clients without first establishing a connection.

joristork, 2021-05-14:
Thank you @dan-sel and @Randelung , I will test your suggestions and feed back.

joristork, 2021-07-07:
I solved / worked around this by simply checking that my client is connected before calling the `disconnect()` method.

joristork, 2021-07-07:
My criterion for checking it is connected:
```python
if (                                                                                                                                                                                           
    hasattr(self._client, "uaclient")                                                                                                                                                          
    and hasattr(self._client.uaclient, "protocol")                                                                                                                                             
    and self._client.uaclient.protocol                                                                                                                                                         
    and (self._client.uaclient.protocol.state != UASocketProtocol.CLOSED)                                                                                                                      
):
```

closed 2021-07-20 by joristork