FreeOpcUa/opcua-asyncio #338: load_enums deprecated, but used in tests

swamper123, 2020-11-23:
https://github.com/FreeOpcUa/opcua-asyncio/blob/d79d204928c1ea06b6bcca1e07e18fcad5a8221b/asyncua/client/client.py#L625

is logged with a warning of being deprecated, but is used in:

https://github.com/FreeOpcUa/opcua-asyncio/blob/08162c339728fd77fa342d37a869a37c3d1db4ca/tests/test_server.py#L530

and

https://github.com/FreeOpcUa/opcua-asyncio/blob/08162c339728fd77fa342d37a869a37c3d1db4ca/tests/test_server.py#L542

The source of this `load_enums()` is in structure and imported in server.py.

But there is also a `load_enums()` in structures104.py, which seems to be just used for `load_data_type_definitions()`. 

Todo:
- replace load_enums from structures.py with load_enums from structure104.py (if necessarry?)
- fix mentioned tests
- link from structures.load_enums() to structures104.load_enums() (or shouldn't calling the method anyway and `load_data_type_definition()` shall be called?)

oroulet, 2020-11-23:
it is not that easy. It is deprecated by spec, but many system on market still use them and our support is not cmplete. So we should have tests for them

swamper123, 2020-11-24:
So enums in 104 are not backwardscompatible to enums in the spec or do I missunderstand something? 
Wouldn't that be a breaking change on OPC side?

So we need Tests anyway with structure 1.04.

oroulet, 2020-11-24:
The specifications of enums is completely different in 1.04.

swamper123, 2020-11-24:
On what version is load_enum based on? 1.03 (since it's the only available one atm)?

I would like to compare them, to understand the things a bit better, what is/shall happening here.

oroulet, 2020-11-24:
as far as I remember load_enum is not necessary anymore. load_data_type_definition() also makes the enums. As far as I know the client code is complete. What is left to do is 1) more testing  2) update the server so it created structs and enum using the new format.
I think I never started 2, but I am not 100% sure... The mianm difference is that the iunforamtion is stored in an attribute called DAtaTypeDefinition or similar see Node.load_data_type_definitions()

oroulet, 2020-11-24:
the new load enum() is inside structres104.py