FreeOpcUa/opcua-asyncio #347: Get event list as client

JoeyFaulkner, 2020-11-25:
Hi there!

Got a bit of what is potentially more of a conceptual issue. I am currently trying to detect events from a Beckhoff PLC's OPCUA-server, with a asyncua client. Firstly, to show what I'm after: using prosys OPCUA client, i can get a list of events that have happened like so:
[image]

and I would like to have that event list available in a python asyncua client script. So, I wrote something like:

```
import asyncio
import logging
from asyncua import Client
from datetime import datetime

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger(__name__)

class SubHandler:
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """
    def event_notification(self, event):
        _logger.info("New event received: %r", event)

async def main():
    url = "opc.tcp://{IP}:4840"
    async with Client(url=url) as client:
        msclt = SubHandler()
        sub = await client.create_subscription(50, msclt)
        handle = await sub.subscribe_events(client.nodes.server)
        await asyncio.sleep(100)
```

And nothing is received. my guess is that the subscription here only responds to *new* events and not existing ones which are still unacknowledged: 
a) is that correct?
b) is there other functionality to get unacknowledged or still active events?

Any help would be greatly appreciated, i hope you're all having a good day!

AndreasHeine, 2020-11-25:
there was something in spec. give me a sec or two

AndreasHeine, 2020-11-25:
in A&C is a ConditionRefresh Method but A&C is WIP...
https://reference.opcfoundation.org/v104/Core/docs/Part9/5.5.7/

there was  also something in the subscription with sending actual values after subscribing if i remember right there are some configurations which are not accessible in the high level client for subscription and monitored items

this code works with the prosys sim server:
```python

import asyncio
import logging
from asyncua import Client
from datetime import datetime

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger(__name__)

class SubHandler:
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """
    def event_notification(self, event):
        _logger.info("New event received: %r", event)

async def main():
    url = "opc.tcp://127.0.0.1:4840"
    async with Client(url=url) as client:
        msclt = SubHandler()
        sub = await client.create_subscription(50, msclt)
        handle = await sub.subscribe_events() #<------ https://github.com/FreeOpcUa/opcua-asyncio/blob/e9cfc5fbc0c1cfdcf5a718387396ca1305a69724/asyncua/common/subscription.py#L183
        while 1:
            await asyncio.sleep(0)

if __name__ == "__main__":
    asyncio.run(main())
```

JoeyFaulkner, 2020-11-26:
You're right, I can also get this working with the prosys server.. this means it must be something to do with the PLC. Thanks so much @AndreasHeine !

closed 2020-11-26 by JoeyFaulkner

AndreasHeine, 2020-11-26:
always a pleasure ;)