FreeOpcUa/opcua-asyncio #349: Connecting to server on Microsoft SQL Server Management Studio

gaurav031295, 2020-12-01:
I have a server running on the Microsoft SQL Server Management Studio. Does anyone know if I can connect to this server as a client using the asyncua package ? The client code specifies the url with the IP address and port number. I tried to enter that by checking the Server Configuration Manager. However, I get a timeout error. So the client never connects.

I think I am getting the concept wrong because I am not sure if it is possible to connect to a server in Server Management Studio as a client. I can successfully connect to the server and write queries with the pyodbc package.

Please Advise!

oroulet, 2020-12-01:
has that server an opcua interface? azure has something,,,

gaurav031295, 2020-12-01:
That is exactly something I am not aware of! I am also not sure how to check it!

swamper123, 2020-12-01:
Well either you have to give a call to the people who provide that SQL Server (Support Hotline or similar) or make your own researches to that topic.

Since our Client works with a bunch of the OPC UA Servers, it may be a reason that this program does not support OPC UA.

closed 2020-12-09 by oroulet