FreeOpcUa/opcua-asyncio #351: asyncua.ua.uaerrors._auto.BadTooManySessions: "The server has reached its maximum number of sessions."

rkrell, 2020-12-01:
After some longer time of operation I get the error below.
What can be done here?

Code:
```python
    url = f'opc.tcp://{cfg.endpoint}:{cfg.port}/'
    async with Client(url=url) as client:
        uri = 'http://Interface_1'
        idx = await client.get_namespace_index(uri)

        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1",
             f"{idx}:Gate"])
        dv = ua.DataValue(ua.Variant(gate, ua.VariantType.Int16))
        await var.write_value(dv)  # set node value using implicit data type

        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1", 
             f"{idx}:Parcel_Barcode"])
        val = '---'
        dv = ua.DataValue(ua.Variant(val, ua.VariantType.String))
        await var.write_value(dv)  # set node value using implicit data type

        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1", 
             f"{idx}:Customer_Reference"])
        val = '---'
        dv = ua.DataValue(ua.Variant(val, ua.VariantType.String))
        await var.write_value(dv)  # set node value using implicit data type

        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1", 
             f"{idx}:Carrier"])
        val = '---'
        dv = ua.DataValue(ua.Variant(val, ua.VariantType.String))
        await var.write_value(dv)  # set node value using implicit data type

        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1", 
             f"{idx}:New_Data"])
        dv = ua.DataValue(ua.Variant(True, ua.VariantType.Boolean))
        await var.write_value(dv)  # set node value using implicit data type

        await asyncio.sleep(0.1)
        var = await client.nodes.root.get_child(
            ["0:Objects", "3:ServerInterfaces", f"{idx}:Interface_1", 
             f"{idx}:OK"])
        while await var.read_value() is False:
            await asyncio.sleep(0.4)
```

```
Traceback (most recent call last):
  File "/home/user/automation/packing/tasks/final.py", line 169, in answer
    await _send_opc_ua(gate, parcel, shipping_information)
  File "/home/user/automation/packing/tasks/final.py", line 76, in _send_opc_ua
    async with Client(url=url) as client:
  File "/home/user/lib/python3.7/site-packages/asyncua/client/client.py", line 72, in __aenter__
    await self.connect()
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/client/client.py", line 245, in connect
    await self.create_session()
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/client/client.py", line 368, in create_session
    response = await self.uaclient.create_session(params)
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 298, in create_session
    data = await self.protocol.send_request(request)
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 151, in send_request
    self.check_answer(data, f" in response to {request.__class__.__name__}")
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 160, in check_answer
    hdr.ServiceResult.check()
  File "/home/user/automation/lib/python3.7/site-packages/asyncua/ua/uatypes.py", line 220, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadTooManySessions: "The server has reached its maximum number of sessions."(BadTooManySessions)
```

AndreasHeine, 2020-12-01:
it depends!
you can try to reduce servers sessiontimout

swamper123, 2021-04-26:
@rkrell Is this still a thing or can it be closed?

rkrell, 2021-04-26:
> @rkrell Is this still a thing or can it be closed?

Not really an issue, happened just one time to me, without changing server parameters.
Closing because there haven't been any more news.
Thanks.

closed 2021-04-26 by rkrell