FreeOpcUa/opcua-asyncio #353: xmlparser - requiredModel not handled

swamper123, 2020-12-07:
Hello there,

the element "RequiredModel" in some xml files (e.g. machinery) is not handled.
This causes crashes, if a special order while importing was not adhered. (import DI.xml -> import Machinery.xml ✔️ ; import Machinery.xml -> import DI.xml ❌ )

Some solution inspirations:
- import just fails, but with error that schema bar requires  schema foo first. (would be quick, but dirty and more helpful)
- a list of all wanted schemas has to be passed to xmlparser with their path, instead of a single string with a path. Then the order will be figured out by that tag and the nodes imported in a working order (if a required schema is not in that list, it shall be tried to find it in the schema folder, otherwise fail)

AndreasHeine, 2020-12-10:
in the machinery nodeset is a required field which tells you what is needed

```xml
  <Models>
    <Model ModelUri="http://opcfoundation.org/UA/Machinery/" Version="1.0.0" PublicationDate="2020-09-25T00:00:00Z">
      <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04.7" PublicationDate="2020-07-15T00:00:00Z" />
      <RequiredModel ModelUri="http://opcfoundation.org/UA/DI/" Version="1.02.2" PublicationDate="2020-06-02T00:00:00Z" />
    </Model>
  </Models>
```

so probably we neet to check that before start parsing!?

oroulet, 2020-12-10:
but how can we know, if DI is required that  Di is already imported? Does not seem to be a mechanism for that.
But maybe we can simply require that all required files are in the same directory..

swamper123, 2020-12-11:
@oroulet You can request: Objects->Server->Namespaces and each object there has mandatory attributes NamespaceUri, NamespaceVersion and NamespacePublicationDate (and some more...). So getting the existing Informations is not a big thing.

I like the folder idea, perhaps a better idea than passing a list of filepaths....but I may start with the "is on the server, else throw Error" solution and open a new issue for a more complex/better way (perhaps with the folder solution).

swamper123, 2021-01-06:
Requirement imports should now be checked. Case closed 🕵️

closed 2021-01-06 by swamper123