FreeOpcUa/opcua-asyncio #360: Getting "InvalidSignature" error when renewing secure channel when working with secure connexion and high server load

kaeros, 2020-12-11:
We might have found a problem on signature validation when working with secure connexions.

Issue occurs when the asyncua is connected with a security_policy equal to : Basic256Sha256.

When the server is facing high load and/or multiple simultaneous security channel renewal, an exception is raised in the verify method at line 375: (since Verifier is an abstract class, it is the VerifierHMac256 class which is triggered with this error)

  File "asyncua\client\ua_client.py", line 82, in _process_received_data
  File "asyncua\common\connection.py", line 333, in receive_from_header_and_body
  File "asyncua\common\connection.py", line 59, in from_header_and_body
  File "asyncua\crypto\security_policies.py", line 162, in verify
  File "asyncua\crypto\security_policies.py", line 375, in verify
  cryptography.exceptions.InvalidSignature

For info this is the code of the culprit: 

[image]

What we can see after investigating in this method is that there is a desynchronisation of signature when the check is performed during channel renewal. So for us there is 2 possible route causes for this to happen:

- The wrong signature which is invalidated by the method is a remnant of a previous signature which have not been updated
- The wrong signature which is invalidated by the method is a future signature which have not been updated yet in the "expected" variable

The thing is, unlike other opcua clients, there is no "retry" alogorithm if an Invalid Signature is detected. So in our case the code will automatically close the concerned subscription if that exception is raised. 

So my question is : do you have any idea why we are facing a signature desynchronisation during secure channel renewal ? And, if not, do you think a "retry behaviour" could be implemented on this specific case to be more robust in case of desynchronisation on signature verification ? Maybe one error could be raised as a warning but 2/3 multiple consecutive bad signature check will raise the InvalidSignature error which then close the subscription ? Maybe implement some kind of "current" and "pending" system on signature (quite like it is done for security tokens)

AsyncIO methodology is not yet my forte and the same thing could be said for openSSL's related subjects so I am open for suggestions to implement a retry algorithm or, even better, for a fix on that desynchronisation issue. 
If needed I can provide multiple screenshot of actual signature desynchronisation with their values (expected and tested). Not sure how it can be used since the data is encrypted and not related to a specific timestamp though. 

Thank you in advance for your time

Best regards

swamper123, 2020-12-14:
> do you have any idea why we are facing a signature desynchronisation during secure channel renewal ? 

You said, that this just happens, if the server is heavy used. I could imagine, that the running eventloop may cause the problem then (especially since en- and decryption needs even more time).

> And, if not, do you think a "retry behaviour" could be implemented on this specific case to be more robust in case of desynchronisation on signature verification ?

I will take a look what an expected behaciour would be. Afaik if a Client fails to renew the lifetime, the lifetime expires and the Subscription is closed on server side... I try to find the source of that.

>  Maybe implement some kind of "current" and "pending" system on signature (quite like it is done for security tokens)

If the signature was not verified when needed, we should ignore/reject them. The security in that case is more important than the subscription in my opinion. The tokens have a small time frame where they are allowed to contain the old Token. I don't know if that is the case for signatures.

kaeros, 2020-12-15:
Thank you for your answer

Concerning the source of the problem, server load do not seems to be the only reason for this to happen. Like I explained, signature desynchronisation might also occurs when you are only working with two clients (from our tests, it seems that simultaneous renewal are also triggering this error). 

I will wait for the result of your investigations for the two other points that you have brought up in your last response.

swamper123, 2020-12-16:
> The thing is, unlike other opcua clients, there is no "retry" alogorithm if an Invalid Signature is detected. So in our case the code will automatically close the concerned subscription if that exception is raised.

[From Part1- 7.11](https://reference.opcfoundation.org/v104/Core/docs/Part1/7.11/):
To protect against non-use by Clients, Subscriptions have a configured lifetime that Clients periodically renew. If any Client fails to renew the lifetime, the lifetime expires and the Subscription is closed by the Server. When a Subscription is closed, all MonitoredItems assigned to the Subscription are deleted.

So if the channel renews and it is failed, a close of the Subscription is wanted behaviour. So no retrys are wanted.
Counter meassurements (well at least the specs mention them) are, that Subs can be reassigned to another Client before the lifetime expires (I think we haven't got such a feature atm?!).

[From Part 6 - 6.7.4:](https://reference.opcfoundation.org/v104/Core/docs/Part6/6.7.4/)
The Server shall reject renew requests if the SenderCertificate is not the same as the one used to create the SecureChannel or if there is a problem decrypting or verifying the signature. T

So the behaviour is wanted and , so I understand that, you need to trigger the whole connection cycle again.

BUT:

> Like I explained, signature desynchronisation might also occurs when you are only working with two clients (from our tests, it seems that simultaneous renewal are also triggering this error).

This sounds still like an issue then. For me it smells like a racecondition thing while processing (which more likely occure, if the server is heavy loaded). :/

kaeros, 2020-12-18:
I might have been confusing when I was talking about 'retries'. I was proposing a retry on the "verify" method that checks if the signature is correct. I was not proposing a retry on the full renew loop if an error is detected. Since we have a desynchronisation in signature checks on renews, we could imagine a simple workaround that just check the signature a first time, if it is not valid, another check is performed after a small tempo and that could be enough to get the signature synchronized again ? The reason we have not tried this on our end is we do not have enough knowledge of asyncIO and asyncUA to avoid a possible break in the asynchronous algorithm ...

A "racecondition issue" ? So that could be linked to multiple concurrent threads that might generate this issue on signature verification ? Let me know if you found anything related to this rootcause, I am very interested in a fix and that is one promising lead in my opinion. Thank you for your work!

sunilgoothy, 2020-12-28:
Same error faced by us. At present, we are using Python OPCUA client with Kepware UA server.
A test program writes to two tags every 1 second.
This Invalid signature error happens every few minutes

Exception raised while parsing message from server
Traceback (most recent call last):
  File "C:\venv\lib\site-packages\asyncua\client\ua_client.py", line 80, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "C:\venv\lib\site-packages\asyncua\common\connection.py", line 329, in receive_from_header_and_body
    chunk = MessageChunk.from_header_and_body(self.security_policy, header, body)
  File "C:\venv\lib\site-packages\asyncua\common\connection.py", line 55, in from_header_and_body
    crypto.verify(header_to_binary(obj.MessageHeader) + struct_to_binary(obj.SecurityHeader) + decrypted, signature)
  File "C:\venv\lib\site-packages\asyncua\crypto\security_policies.py", line 162, in verify
    self.Verifier.verify(data, sig)
  File "C:\venv\lib\site-packages\asyncua\crypto\security_policies.py", line 375, in verify
    raise uacrypto.InvalidSignature
cryptography.exceptions.InvalidSignature

Whenever, this happens, our client program closes connection and re-establishes again.
But this seems to be excessive to do every few minutes.

swamper123, 2021-01-04:
> I was proposing a retry on the "verify" method that checks if the signature is correct. 

Yes, I understood that. But you said, that this happenes when the Server is in channel renewing.

And the specs say:

> The Server shall reject renew requests if the SenderCertificate is not the same as the one used to create the SecureChannel or if there is a problem decrypting or verifying the signature. The Client shall abandon the SecureChannel if the Certificate used to sign the response is not the same as the Certificate used to encrypt the request

So this is still wanted behaviour. What is not written inside is, HOW the signature is verified, that is true...but I am not sure how we could implement such a thing in a nice way at the moment.

Race conditions are more rare in asyncio than in threads, but can still appear (mostly a reason of mysterious bugs and not easy to recreate). I could imagine that a desync may appear, because of that renewal. We can't determine in the loop when this may be done. 
We would need to tell the eventloop "try it again next iter again" somehow/somewhere and catching the error and counting the retries.

AndreasHeine, 2021-01-04:
if it happens only on high serverload you might think about the servercapabilities each hardware has limits especialy with encryption which cost performance and its still python and not c

what are the dimensions of the server?
how many nodes? do they change faster then 10ms?
how many clientconnections?
how many subscriptions?
Historizing?
Events fired?

do you block somewhere the eventloop like in a subhandler?

@sunilgoothy which kepserver version? i am at 6.9.572.0

sunilgoothy, 2021-01-05:
> if it happens only on high serverload you might think about the servercapabilities each hardware has limits especialy with encryption which cost performance and its still python and not c
>
> what are [...]

I have latest release of Kepware V6.9.584.0

Server load is very thin. It is used only to test python client.
Python Client subscribes to 100 tags. Two tags values are written every second.
Only one client is connected to Kepware server.
No history or events used. SubHandler is used to handle subscription updates. Again only subscription updates client it will be receiving is the for two tags it writes to.
Python Client keeps getting "Invalid Signature" error on an average every 30 minutes (Sometime its within 10 min and sometime it takes an hour. But never seen more than hour)

kaeros, 2021-01-06:
> We would need to tell the eventloop "try it again next iter again" somehow/somewhere and catching the error and counting the retries.

Yes that is what I was proposing. It could give us a workaround if the InvalidSignature error is due to a desynhronisation between the received message's signature and the expected one in the check. A small retry/delay could fix the problem since the error is ponctual and the system have been found to pass multiple renews before that error occurs.

> what are the dimensions of the server?
how many nodes? do they change faster then 10ms?
how many clientconnections?
how many subscriptions?
Historizing?
Events fired?

As said before, this error has occured using several server configurations. The key point is to have multiple channel renewal at the same time (or during the same laps of time it takes to perform one renew).
That being said, our serveur's nominal configuration is 10 sessions with 50k monitored nodes that are read every 1s (and are incremented pretty much at the same period 1s/2s). But we managed to reproduce the issue with less clients (sessions) and less monitored items. No particular history or events are used and an handler is used to handle subscription (like it is used in examples files).

AndreasHeine, 2021-01-06:
i assume we should need to wait for a no communication gap to renew the session and while renewing whe should queue the requests till the renewing is done !?

AndreasHeine, 2021-01-07:
> But I fear that "periodic renews" are a must have in an "opcua compliant library" (mandatory operations described in the protocol documentation ?)

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/

**requestedLifetime | Duration | The requested lifetime, in milliseconds, for the new SecurityToken. It specifies when the Client expects to renew the SecureChannel by calling the OpenSecureChannel Service again. If a SecureChannel is not renewed, then all Messages sent using the current SecurityTokens shall be rejected by the receiver.Several  cryptanalytic attacks become easier as more material encrypted with a  specific key is available. By limiting the amount of data processed  using a particular key, those attacks are made more difficult. Therefore  the volume of data exchanged between Client and Server must be limited by establishing a new SecurityToken after the lifetime.The  setting of the requested lifetime depends on the expected number of  exchanged messages and their size in the lifetime. A higher volume of  data requires shorter lifetime.**

the thing is from client side: if we make a renew we should not do that if a request ist not reponded and we should not send a request if we want to renew and we have 30% of the SecureChannel lifetime for that

kaeros, 2021-01-08:
> the thing is from client side: if we make a renew we should not do that if a request ist not reponded and we should not send a request if we want to renew and we have 30% of the SecureChannel lifetime for that

Is that feature already implemented in async-opcua's client ? For what I could see in the code, as soon as the "0.7 * channel lifetime" threshold (the 30%) is reached, the renew is triggered regardless of pending operations. Is this an enhancement already planned in the backlog ?
Just to sum up, basically the idea you are proposing is : making sure that all pending requests are processed and then queuing (postponing) incoming requests during channel renewal, in order to avoid signature desynchronisation?

AndreasHeine, 2021-01-08:
> Is that feature already implemented in async-opcua's client ? For what I could see in the code, as soon as the "0.7 * channel lifetime" threshold (the 30%) is reached, the renew is triggered regardless of pending operations. Is this an enhancement already planned in the backlog ?

planned: i dont think so (maybe the other guys did) but i have seen some of these issues in the past year so its getting more and more clear thats it is a bug

> Just to sum up, basically the idea you are proposing is : making sure that all pending requests are processed and then queuing (postponing) incoming requests during channel renewal, in order to avoid signature desynchronisation?

exacly! but its more like loud thinking...

oroulet, 2021-01-08:
I did not follow the discussion closely but can't we keep the old signature for some time after renewal like we do with tokens?

AndreasHeine, 2021-01-08:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/

2. and 3.

AndreasHeine, 2021-01-09:
https://github.com/FreeOpcUa/opcua-asyncio/blob/d3c39afc5fd59613c5b273a5819e626baf1f6f42/asyncua/common/connection.py#L144

oroulet, 2021-01-10:
What is the point of your links? I know we keep last tokenid

AndreasHeine, 2021-01-10:
@oroulet sorry for the delay my laptop battery died yesterday ...

can enyone verify that this solves the issue:

my testsetup is python client against kepserverex6.

```python
import asyncio, logging, datetime
from asyncua import Client, ua
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger("asyncua")

cert = "cert.pem"
private_key = "key.pem"
node = "ns=2;s=Beispiele für Simulation.Funktionen.Ramp1"

async def main():
    client = Client("opc.tcp://127.0.0.1:49320")
    client.secure_channel_timeout = 5*1000
    client.session_timeout = 5*1000
    client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        certificate=cert,
        private_key=private_key,
    )
    async with client:
        print(datetime.datetime.now())
        while 1:
            await asyncio.sleep(.1)
            # print(datetime.datetime.now())
            print(await client.get_node(node).read_value())

if __name__ == "__main__":
    asyncio.run(main())
```

i was able to get consistent invalid signature (so safely reproducible within 30s)
after following changes it runs smooth since 15min

```python
async def _renew_channel_loop(self):
        """
        Renew the SecureChannel before the SessionTimeout will happen.
        In theory we could do that only if no session activity
        but it does not cost much..
        """
        try:
            duration = min(self.session_timeout, self.secure_channel_timeout) * 0.7 / 1000
            while True:
                # 0.7 is from spec. 0.001 is because asyncio.sleep expects time in seconds
                await asyncio.sleep(duration)
                _logger.debug("renewing channel")
                await self.open_secure_channel(renew=True)
                # val = await self.nodes.server_state.read_value() # <----------------- commented out
                # _logger.debug("server state is: %s ", val) # <----------------- commented out
        except asyncio.CancelledError:
            pass
        except Exception:
            _logger.exception("Error while renewing session")
            raise
```

oroulet, 2021-01-10:
Why would commenting these lines change anything?the session is already renewed.

If that helps it must be some side effects of some race conditions...

AndreasHeine, 2021-01-10:
yes i agree! but it appears not to happen if the read_value is commented out. I just start to peal the onion... ^^

kaeros, 2021-01-12:
Unfortunately on our end, commenting these two lines did not prevent the InvalidSignature error. 
Since this error appereances are sparse, we are continuing our tests to see if there is an enhancement with these changes in the error's occuring rate.

AndreasHeine, 2021-01-12:
@kaeros do you use python client and server or is the server another stack than python?

kaeros, 2021-01-12:
@AndreasHeine We are using the python library only client side. Server side we are using a C "High Performance" opcua stack from Unified Automation.

ZuZuD, 2021-01-18:
This [patch](https://github.com/FreeOpcUa/opcua-asyncio/compare/master...ZuZuD:invalid_signature) should cover the racy secure_channel renewal issue. In a nutshell, it keeps the last secure_channel key (symmetric) in memory, and tries to decrypt/verify the signature with it on `InvalidSignature` exceptions. Also for security concerns, it removes the keys from memory as soon as it has expired.

If some of you could try and give a feedback it would be much appreciated. 
I was able to repro with the following setup:
```
./uaserver --certificate ../examples/certificate-example.der --private_key ../examples/private-key-example.pem -v DEBUG
```
and client configuration from IPython:
```
import logging
import asyncio
import datetime
import sys;
sys.path.insert(0, "/home/user/github/opcua-asyncio");
import asyncua;
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256, SecurityPolicyBasic256, SecurityPolicyBasic128Rsa15
import cryptography
from asyncua.crypto import uacrypto
from asyncua.common.shortcuts import Shortcuts
logger = logging.getLogger("asyncua")
logger.setLevel(logging.WARN)
handler = logging.StreamHandler(sys.stdout)
formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
handler.setFormatter(formatter)
logger.addHandler(handler)
pk = "/home/user/github/opcua-asyncio/examples/certificates/peer-private-key-example-1.pem"
cert = "/home/user/github/opcua-asyncio/examples/certificates/peer-certificate-example-1.der"
client = asyncua.Client("opc.tcp://127.0.0.1:4840")
client.secure_channel_timeout = 1*1000
client.session_timeout = 1*1000
s = Shortcuts("opc.tcp://127.0.0.1:4840")
node = s.service_level
await client.set_security(SecurityPolicyBasic256Sha256,certificate=cert,private_key=pk)
async def run():
    await client.connect()
    print(f"start: {datetime.datetime.now()}")
    cont = True
    while cont:
        await asyncio.sleep(0.01)
        try:
            slevel = await client.get_node(node).read_value()
        except Exception as e:
           cont = False
           await client.disconnect()
           print(f"end Exception: {datetime.datetime.now()} with excep: {e}")
await run()
```

sunilgoothy, 2021-01-20:
> 
> 
> This [patch](https://github.com/FreeOpcUa/opcua-asyncio/compare/master...ZuZuD:invalid_signature) should cover the racy secure_channel renewal issue. In a nutshell, it keeps the last secure_channel key (symmetric) in memory, and tries to decrypt/verify the signature with it on `InvalidSignature` exceptions. Also for security concerns, it removes the keys from memory as soon as they have expired.
> 
> If some of you could try it and provide a feedback it would be much appreciated.

This patch is working. I am running client now for almost 24hours and there was no error during this time.
I will update if there is any issue.

Thanks for the patch.

kaeros, 2021-01-21:
Sorry for the late response.

We tested the patch on our end and the error is effectively bypassed by the fix on both our systems (both stack previously mentioned). I know that this is not the solution we were discussing (queuing operations while perorming channels renewals) but it seems that keeping the prev signature, to validate "invalidSignature" cases which may be detected, is a good workaround.

I cannot say though if the fix is compliant with the opc foundation documentation on "channel renewals" but thank you for the workaround!

ZuZuD, 2021-01-21:
Thanks for the feedback.

closed 2021-01-29 by kaeros