FreeOpcUa/opcua-asyncio #390: Ha tests randomly failing

oroulet, 2021-01-06:
elf = <tests.test_ha_client.TestHaClient object at 0x7f60377cd7d0>
ha_config = HaConfig(ha_mode=<HaMode.WARM: 1>, keepalive_timer=1, manager_timer=1, reconciliator_timer=1, session_timeout=30, request_timeout=30, session_name='HaClient', urls=['opc.tcp://127.0.0.1:40489', 'opc.tcp://127.0.0.1:47553'])
ha_servers = (OPC UA Server(opc.tcp://127.0.0.1:40489), OPC UA Server(opc.tcp://127.0.0.1:47553))
client_key_and_cert = ('/home/runner/work/opcua-asyncio/opcua-asyncio/examples/certificates/peer-private-key-example-1.pem', '/home/runner/work/opcua-asyncio/opcua-asyncio/examples/certificates/peer-certificate-example-1.der')

    @pytest.mark.asyncio
    async def test_security(self, ha_config, ha_servers, client_key_and_cert):
        # check security can be set via method and constructor
        ha_clients = []
        srv1, srv2 = ha_servers
        srv1_port = srv1.endpoint.port
        srv2_port = srv2.endpoint.port
    
        ha_config = ha_config(srv1_port, srv2_port)
        ha_c1 = HaClient(ha_config)
        key, cert = client_key_and_cert
    
        security_policy = security_policies.SecurityPolicyBasic256Sha256
        user_key = CertProperties(key, "PEM")
        user_cert = CertProperties(cert, "DER")
        security_mode = ua.MessageSecurityMode.SignAndEncrypt
        ha_c1.set_security(security_policy, user_cert, user_key, mode=security_mode)
        ha_clients.append(ha_c1)
    
        ha_security = HaSecurityConfig(
            security_policy, user_cert, user_key, mode=security_mode
        )
        ha_clients.append(HaClient(ha_config, ha_security))
    
        for ha_client in ha_clients:
            assert ha_client.security_config.policy == security_policy
            assert ha_client.security_config.mode == security_mode
            assert ha_client.security_config.certificate == user_cert
            assert ha_client.security_config.private_key == user_key
    
            await ha_client.start()
            await wait_clients_socket(ha_client, UASocketProtocol.OPEN)
    
            for client in ha_client.get_clients():
                assert isinstance(client.security_policy, ua.SecurityPolicy)
                assert isinstance(client.uaclient.security_policy, ua.SecurityPolicy)
>               assert client.security_policy.Mode == security_mode
E               assert <MessageSecurityMode.None_: 1> == <MessageSecur...AndEncrypt: 3>
E                 +<MessageSecurityMode.None_: 1>
E                 -<MessageSecurityMode.SignAndEncrypt: 3>

tests/test_ha_client.py:252: AssertionError

tht test sometimes fails.

oroulet, 2021-01-06:
@ZuZuD  you are the only one understanding that stuff

ZuZuD, 2021-01-06:
> @ZuZuD you are the only one understanding that stuff

Let me have a look.

ZuZuD, 2021-01-06:
`for i in $(seq 1 100); do pytest -W ignore::DeprecationWarning tests/test_ha_client.py; done` gives me 100% success with various python versions. @oroulet  can you provide more details about the environment throwing those errors?

ZuZuD, 2021-04-26:
#552 probably addresses this, closing.

closed 2021-04-26 by ZuZuD