FreeOpcUa/opcua-asyncio #395: TCP keepalive on server side

nickjhall, 2021-01-12:
This is more of a question than an issue.

Is there currently a way to set TCP keepalive on the server socket?

I haven't tried it yet, but it [looks to be possible](https://stackoverflow.com/questions/12248132/how-to-change-tcp-keepalive-timer-using-python-script) to do this after the socket it opened. But I think it would be nicer if it was integrated into the server config.

Best regards,
Nick

AndreasHeine, 2021-01-12:
Questions -> https://github.com/FreeOpcUa/opcua-asyncio/discussions
;)

nickjhall, 2021-01-12:
Ok then, I'll move it to here: [https://github.com/FreeOpcUa/opcua-asyncio/discussions/396](https://github.com/FreeOpcUa/opcua-asyncio/discussions/396)

closed 2021-01-12 by nickjhall