FreeOpcUa/opcua-asyncio #397: Server doesn't satisfy required XML-Models. Import them first!

pbertoni89, 2021-01-13:
I was working correctly with [0.9.3](https://github.com/FreeOpcUa/opcua-asyncio/releases/tag/0.9.3) from pip and PackML nodeset from [UA-Nodeset v1.04](https://github.com/OPCFoundation/UA-Nodeset).

Casual upgrade to 0.9.10 (due to docker image re-build) broke the xml import process with

```
 File "<myscript>.py", line 210, in init
    await self.s.import_xml('<mypath>/UA-Nodeset/PackML/Opc.Ua.PackML.NodeSet2.xml')
  File "<myenv>/asyncua/server/server.py", line 543, in import_xml
    return await importer.import_xml(path, xmlstring)
  File "<myenv>/asyncua/common/xmlimporter.py", line 91, in import_xml
    await self._check_required_models(xmlpath, xmlstring)
  File "<myenv>/asyncua/common/xmlimporter.py", line 80, in _check_required_models
    raise ValueError("Server doesn't satisfy required XML-Models. Import them first!")
ValueError: Server doesn't satisfy required XML-Models. Import them first!
```

pbertoni89, 2021-01-13:
0.9.8 is **not** affected, whereas 0.9.9 raises the same error

oroulet, 2021-01-13:
@swamper123 looks like an issue here. Also aren't you supposed to list the missing models in the error message? we really need to do that

pbertoni89, 2021-01-13:
Sure, exact file is under the tree of the given link to v1.04: [Opc.Ua.PackML.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PackML/Opc.Ua.PackML.NodeSet2.xml).

swamper123, 2021-01-13:
It seems, that you are missing XML models for PackML. With the actual master branch, it should log correctly which XML Versions are missing (@oroulet that commit was done after the release, so you may have to check them on your own).

swamper123, 2021-01-13:
@pbertoni89 
`<RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04.6" PublicationDate="2020-04-14T00:00:00Z" />`

I guess you may have to generate the newest Version of the UA addresspace. Does your Server provide Version 1.04.6 of the UA namespace or higher?

swamper123, 2021-01-13:
If you exclude the xml_import, and ran the blank server, you can check it via:
Root->Objects->Server->Namespaces->http://opcfoundation.org/UA/ -> NamespaceVersion

pbertoni89, 2021-01-13:
Value is `1.04.6`. Till now, just checked cf278b5 from master and import runs smoothly, so I'm closing. Thank you!

closed 2021-01-13 by pbertoni89

pbertoni89, 2021-01-13:
Just an additional note because I misreported NamespaceVersion.
It was `1.04.6` for `http://opcfoundation.org/UA/` object, whereas `http://opcfoundation.org/UA/PackML/` had value `1.01`