FreeOpcUa/opcua-asyncio #431: Server's UserManager interface is unusable when encryption enabled

che2116, 2021-02-02:
When the server uses encryption for data transferring, a password parameter passed to the UserManager.get_user method is encoded and encrypted. So it is impossible to use this data for the user's authentication.

https://reference.opcfoundation.org/v104/Core/docs/Part4/7.36.4/
https://reference.opcfoundation.org/v104/Core/docs/Part4/7.36.2/#7.36.2.2

starturtle, 2021-02-02:
I'm confused. The spec excerpts you shared both demand the password to be encrypted, so that happening is not a mistake, but specified behavior.
At the point the client ist authenticates, the encrypted channel is already established, so it also makes sense that it gets used.
Do you mean to say that it doesn't get decrypted on server side early enough to be usable for authentication?

che2116, 2021-02-03:
> Do you mean to say that it doesn't get decrypted on server side early enough to be usable for authentication?

Exactly