FreeOpcUa/opcua-asyncio #438: Error: Number of acknowledgement results (0) does not match number of acknowledgements (1)

droneconnect, 2021-02-07:
I am new to opcua-asyncio. 
Just installed  it with pip3 install asyncua
I run python 3.8.5 on ubuntu 20.04

I run the https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-minimal.py and getting an error in uaexpert 1.5.1

could anyone advice?
BR
Dan

[image]

yichenj, 2021-02-08:
Same issue. It seems a legacy issue from the sync version. [](https://github.com/FreeOpcUa/python-opcua/issues/803)

AndreasHeine, 2021-02-08:
looks like there is a ack missing... it actually does not do any bad you still get the notifications in ua expert! 

request:

subscription   Acknowledgements [ ] | Subscription Acknowledgement | The list of acknowledgements for one or more Subscriptions. This list may contain multiple acknowledgements for the same Subscription (multiple entries with the same subscriptionId). This structure is defined in-line with the following indented items.

response:

results [ ] | StatusCode | List of results for the acknowledgements (see 7.34 for StatusCode definition). The size and order of the list matches the size and order of the subscriptionAcknowledgements request parameter.

swamper123, 2021-02-08:
I went back with our versions until v0.9.0, but the issue still exist. So the newer commits haven't made any influence on that.

oroulet, 2021-02-08:
What is an acknowledgement result?
And do we do something wrong?

AndreasHeine, 2021-02-08:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.5/

publish response -> results is empty but it must have same number of statuscodes like the number of subscriptionAcknowledgements

oroulet, 2021-02-08:
I had a short look at it and in fact looks like we have a parsing bug and something very strange with the statuscode implementation. There is both an enum and a class. And enum should be more than enough i suppose...
Must look at it later

droneconnect, 2021-02-08:
good stuff! thanks guys for quickly troubleshooting this :).. give us an update when we can test again!

AndreasHeine, 2021-02-08:
you can use it anyways that bug does not break anything you will recv all notifications in UA Expert! just kind of ugly in the log...

yichenj, 2021-02-09:
Sorry, is this the same issue with [the sync version](https://github.com/FreeOpcUa/python-opcua/issues/803)?
The sync one looks very complicated and can cause severe result based on the implementation of client.

oroulet, 2021-02-09:
@AndreasHeine @swamper123 I tried to do something yesterday: Send results immediatly after receiving acks (2 acks in PublishRequest-> 2 statuscode results in next PublishResponse) But it does not seem to make uaexpert happy. I am not really sure what it wants...

oroulet, 2021-02-09:
I am wondering if uaexpert expects the status code in the PublishResponse corresponding to the PublishRequest, but that would be orthogonal to the entire pub/sub system.... We keep a queue/buffer of PublishRequests in server

AndreasHeine, 2021-02-09:
@oroulet this is from another server with node-opcua stack:
[image]
[image]
[trace1.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/5950837/trace1.zip)

brubbel, 2021-02-09:
> I went back with our versions until v0.9.0, but the issue still exist. So the newer commits haven't made any influence on that.
That is because UaExpert only started reporting this in the logs in the more recent versions.

Don't try to decode my answer below word by word (it's been a long time for me), but basically it goes as follows.

1. client sends publication request, which acts a 'token' for the server to send a future datachange.
2. server sends datachange notification (consumes token), pushes a 'pending ack' in an ordered list.
3. client sends new publication request (thus a new token), with embedded ack for notification in (2)
Now comes the issue:
4. server sends new datachange notification, with resulting statuscode 'good' after processing the client's ack in (3), then server removes that ack from the list mentioned in (2)

So the idea is here:
* clients sends pub requests. Each request can be used by the server as a token (permission) to send new datachange events.
* server has no tokens left -> stops sending datachange events (1 request = 1 response)

Freeopcua does not implement the token/acking whatever thing.
Current implementation of freeopcua is to send datachange events when it feels to (=wrong!)

* At connection/subscription time, the client sends a few requests (tokens) to the server so that the above procedure can be initiated. If I remember correctly (50% chance), 2 requests are queued by uaexpert at any time (but this number can be higher, to handle higher latency for example).

samxu0823, 2021-09-06:
I have met the same problem with opcua 0.98.13 and UaExpert1.5.1.
Is there a solution right now? Thank you and looking forward to your answer!

AndreasHeine, 2021-09-06:
any exception or error? or just the log in uaexpert?

samxu0823, 2021-09-06:
only log in UaExpert

AndreasHeine, 2021-09-07:
do not worry!

Soeren1987, 2022-09-19:
Is there any updates on this ??