FreeOpcUa/opcua-asyncio #457: Exception in subscription loop: KEPServerEX 6.9 opc ua client

AndreasHeine, 2021-02-25:
**Trace**<br />

> ERROR:asyncua.server.internal_subscription:Exception in subscription loop
> Traceback (most recent call last):
>   File "asyncua\server\internal_subscription.py", line 76, in _subscription_loop
>   File "asyncua\server\internal_subscription.py", line 113, in publish_results
>   File "asyncua\server\uaprocessor.py", line 87, in forward_publish_response
>   File "asyncua\server\uaprocessor.py", line 46, in send_response
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 271, in to_binary
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 271, in to_binary
>   File "asyncua\ua\ua_binary.py", line 243, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 283, in list_to_binary
>   File "asyncua\ua\ua_binary.py", line 283, in <listcomp>
>   File "asyncua\ua\ua_binary.py", line 261, in to_binary
>   File "asyncua\ua\ua_binary.py", line 175, in pack_uatype
>   File "asyncua\ua\ua_binary.py", line 446, in extensionobject_to_binary
>   File "asyncua\ua\ua_binary.py", line 243, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 283, in list_to_binary
>   File "asyncua\ua\ua_binary.py", line 283, in <listcomp>
>   File "asyncua\ua\ua_binary.py", line 271, in to_binary
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 261, in to_binary
>   File "asyncua\ua\ua_binary.py", line 181, in pack_uatype
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 261, in to_binary
>   File "asyncua\ua\ua_binary.py", line 179, in pack_uatype
>   File "asyncua\ua\ua_binary.py", line 364, in variant_to_binary
>   File "asyncua\ua\ua_binary.py", line 175, in pack_uatype
>   File "asyncua\ua\ua_binary.py", line 446, in extensionobject_to_binary
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 271, in to_binary
>   File "asyncua\ua\ua_binary.py", line 248, in struct_to_binary
>   File "asyncua\ua\ua_binary.py", line 261, in to_binary
>   File "asyncua\ua\ua_binary.py", line 171, in pack_uatype
>   File "asyncua\ua\ua_binary.py", line 35, in pack
>   File "asyncua\ua\uatypes.py", line 40, in datetime_to_win_epoch
> AttributeError: 'str' object has no attribute 'tzinfo'

**Version**<br />
Python-Version: 3.6 64bit
opcua-asyncio Version (e.g. master branch, 0.9): v0.9.12

KEPServerEX:
connection localhost (asyncua server) -> KEPServer 2 Tags -> ns=0;i=2267 and ns=0;i=2256 (standard stuff) -> UaExpert (connect to KEPServerEX) -> Sub ns=0;i=2256 -> Exception in asyncua server

AndreasHeine, 2021-02-25:
found the reason why it happens:
https://github.com/FreeOpcUa/opcua-asyncio/blob/f6aa7a972fdc742ff2c983aa1a00437bb359702b/asyncua/server/server.py#L114

https://reference.opcfoundation.org/v104/Core/DataTypes/BuildInfo/

buildDate | UtcTime <-- we propably should make docstrings to hint the types or make some typechecking?

@oroulet if you agree i fix it with a PR?

oroulet, 2021-02-25:
Fine for me

closed 2021-03-05 by AndreasHeine