FreeOpcUa/opcua-asyncio #463: Unable to read all Child nodes and Sub-Child nodes variable data at one shot.

arunkrh123, 2021-03-01:
### **I am able to read the Node and immediate variable data values using the below code.**

async def readValueDirect(NodeName, VarName, SecurityMode='None', UserIdentityTokenType='Anonymous', serverCreds=None,
                          authCredentials=None):
    DISCOVERY_SERVER_ENDPOINT = 'opc.tcp://{}:{}'.format(str(IP), str(PORT))
    client = Client(DISCOVERY_SERVER_ENDPOINT)
    print("Getclient:", await get_client(client, SecurityMode, UserIdentityTokenType, authCredentials, serverCreds))
    try:
        async with client:
            uri = 'http://opcfoundation.org/UA/'
            try:
                idx = await client.get_namespace_index(uri)
            except ua.UaError as exec:
                print(str(exec))
            # Now getting a variable node using its browse path
            CHILDRENS = await client.nodes.objects.get_children()

            for N in CHILDRENS:
                GetN = client.get_node(N)
                node_variables = await GetN.get_variables()
                for V in node_variables:
                    V1 = await GetN.read_browse_name()
                    V2 = await V.read_browse_name()
                    V1 = str(V1)
                    V2 = str(V2)
                    V1 = V1[16:-1]
                    V2 = V2[16:-1]
                    V3 = str(V1) + "-" + str(V2)
                    NODE = V
                    NODE = str(NODE)
                    NODE = NODE[26:-2]
                    NS = V
                    NS = str(NS)
                    NS = NS[22:-7]
                    PN = await V.get_parent()
                    PN = str(PN)
                    PN = PN[26:-2]
                    Dtype = str(await V.read_data_value())
                    Dtype = re.search('(?<=VariantType.)\w+', Dtype)
                    Dtype = str(Dtype[0])
                    VarValue = await  V.read_value()
                    ObjectDict[V3] = {"GetNName": await GetN.read_browse_name(), "DataType": Dtype,
                                      "NodeBrowseName": await V.read_browse_name(), "Nodes": V,
                                      "PARENTNODE": await V.get_parent(), "DisplayName": await V.read_display_name(),
                                      "NodeID": V.nodeid, "BrowseName": await V.read_browse_name(),
                                      "DataValue": await  V.read_data_value(), "Value": VarValue, "VName": V,
                                      "Namespace": NS, "Index": NODE, "ParentIndex": PN}
            pprint.pprint(ObjectDict)

    except TimeoutError as exec:
        print(exec)
        traceback.print_exc()
        return 0

def ReadNodeValue(NodeName, VarName, SecurityMode='None', UserIdentityTokenType='Anonymous', serverCreds=None,
                  authCredentials=None):
    p = asyncio.run(
        readValueDirect(NodeName, VarName, SecurityMode=SecurityMode, UserIdentityTokenType=UserIdentityTokenType,
                        serverCreds=serverCreds, authCredentials=authCredentials))
    return p

SetClient('192.168.204.7', '4840')  # Successfully establishing the communication
x = ReadNodeValue("Server", "ServerStatus") 

When i use **'CHILDRENS = await client.nodes.objects.get_children()'** in the code I am not getting all Sub-Child Node-ID.

Is there any way I can get all the **CHILD and SUB-CHILD** node variable values into a dictionary?

Please refer the below image for understanding the Node hierarchy.

[image]

Thanks in advance.

Regards,
Arunkumar H

AndreasHeine, 2021-03-01:
https://github.com/FreeOpcUa/opcua-asyncio/blob/12c0e7a05d756219ac56448edf7f78a030a3d049/asyncua/common/node.py#L323

get_children() returns a list of nodes (Node-Class-Instances) like in the docstring:

> Get all children of a node. By default hierarchical references and all node classes are returned.

what childs do you miss? example?
what do you mean with `I am not getting all Sub-Child Node-ID` ???

Koseng, 2021-03-01:
Hi,
get_children() just gives you the next layer. In ua_utils.py there are helper methods to collect all children recursively.

**Example to collect all variables:**
```
from asyncua.common import ua_utils
   
nodeList = await ua_utils.get_node_children(myStartNode)
for n in nodeList:
     nodeClass = await n.read_node_class()
     if nodeClass == ua.NodeClass.Variable:
             # do something
```

arunkrh123, 2021-03-02:
Hi Koseng and AndreasHeine,

Thanks for your response.

I have tried the code which @Koseng shared in the comment section, I am able to read all the child and sub-child nodes. But I am getting error  when I try to read the value using this NodeIDs.

Note: All attributes are not supported by few node variables. Is there any way to identify what attributes are supported by the node variable?

Code:
CHILDRENS = await client.nodes.objects.get_children()
          allVariables = []
          for N in CHILDRENS:
              ######################
              from asyncua.common import ua_utils

              allVariables = []

              nodeList = await ua_utils.get_node_children(N)

              print("test node list: ",nodeList)

              for n in nodeList:
                  nodeClass = await n.read_node_class()
                  if nodeClass == ua.NodeClass.Variable:
                      if nodeClass not in allVariables:
                         allVariables.append(n)
                      #val= await  ua.NodeClass.value

              for nodeID in allVariables:
                  GetNtest = client.get_node(nodeID)
                  Value = await GetNtest.read_value()
                  BrowseName = await GetNtest.read_browse_name()

                  print("Name",BrowseName," Value:",Value)

[image]

Kindly please help me to resolve this issue. My objective is to read all NodeIDs and their corresponding node value to a dictionary as Key value pair.

Kindly please help to resolve this issue.

Thanks and Regards,
Arunkumar H

arunkrh123, 2021-03-02:
Hi @Koseng,

Thanks for your quick response.

I am getting following error when I run the above code. 

[image]

INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.client.ua_client.UaClient:browse
[Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.SerialNumber), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.RevisionCounter), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.Manufacturer), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.Model), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.DeviceManual), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.DeviceRevision), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.SoftwareRevision), Node(ns=4;s=|vprop|PM5630-2ETH_OPC UA.Application.HardwareRevision), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.POU_1.a1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.POU_1.ai1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.CompilerVersion), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.CompilerVersion.uiMajor), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.CompilerVersion.uiMinor), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.CompilerVersion.uiServicePack), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.CompilerVersion.uiPatch), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.RuntimeVersion), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.RuntimeVersion.uiMajor), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.RuntimeVersion.uiMinor), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.RuntimeVersion.uiServicePack), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.Constants.RuntimeVersion.uiPatch), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ai1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ai2), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ai3_1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ai4), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ao1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ao2), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ao3), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.ao4), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI10), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI11), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI12), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI13), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI14), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI15), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI16), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI17), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI18), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI19), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI2), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI20), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI21), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI22), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI23), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI24), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI25), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI26), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI27), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI28), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI29), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI3), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI30), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI31), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI32), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI4), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI5), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI6), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI7), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI8), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.DI9), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do1), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do10), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do11), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do12), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do13), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do14), Node(ns=4;s=|var|PM5630-2ETH_OPC UA.Application.IoConfig_Globals_Mapping.do15), Node(ns=4;s=|var|PM5630-2ETH [... cut ...]
    result[0].StatusCode.check()
  File "C:\Users\arunkumar.h\Desktop\HMProjects\HMAutomation\intsdn01.1\venv\lib\site-packages\asyncua\ua\uatypes.py", line 220, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadDataUnavailable: "Expected data is unavailable for the requested time range due to an un-mounted volume, an off-line archive or tape, or similar reason for temporary unavailability."(BadDataUnavailable)

Process finished with exit code 1

Appreciate your help.

Thanks,
Arunkumar H

omidfathi, 2022-06-29:
Hello again

There is a new problem, to collecting the values an error happened "asyncua.common.utils.NotEnoughData"
[image]

closed 2023-03-29 by schroeder-