FreeOpcUa/opcua-asyncio #465: Servers XML is not valid against UANodeSet.xsd

AndreyKl4, 2021-03-05:
Dear all,

I faced the behavior that i didn't expect. I want describe it in a short:

For my Project i need to load a whole XML(Information Model) of the Server and after that load it in Siemens opcua Client. I did the folowing steps:

1) I created XML-file of Server with line await self.server.export_xml_by_ns("save_parameters_xml_server.xml") (I attached thr XML-File)

2) I tried to download the XML in Siemens and got the error.

3) I used SioME (https://support.industry.siemens.com/cs/document/109755133/siemens-opc-ua-modeling-editor-(siome)-zur-umsetzung-von-opc-ua-companion-spezifikationen?dti=0&lc=de-WW) to test XML-File and got the errors that XML is not valid against UANodeSet.xsd.

4) For sureness i used XML-Validator (https://www.freeformatter.com/xml-validator-xsd.html) to validate my XML against Schema ( UANodeSet.xsd). As a result I got the same errors.

My question is this behavior correct? In my o

pinnion, it looks like a bug. But if the behavior correct can smb tell me please in what way i can create a XML-File of Server that is valid against UANodeSet.xsd?

Thanks in Advance!

Andrii

[XML.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/6089533/XML.txt)

swamper123, 2021-03-05:
Hi @AndreyKl4 ,

it is possible, that the exporter still has issues, especially with more complex expressions.
In your case, all exported UADataType are faulty and one line with Encoding in "InputArguments".

Had you import any required model for the ns?
Can you provide the code (at least the part, where you make the Nodes), how you modeled the ns (that would help me to understand and replicate the stuff faster and better)?

swamper123, 2021-03-05:
Quick-fix: @AndreyKl4 : I made the observation, that for all UADatatype, you have to switch the position in the xml between References and Definition:

Before: 
```xml
  <UADataType NodeId="ns=1;i=7" BrowseName="1:rc_april_tag_detect_detect_args_robot_pose">
    <DisplayName>rc_april_tag_detect_detect_args_robot_pose</DisplayName>
    <Description>rc_april_tag_detect_detect_args_robot_pose</Description>
    <Definition Name="rc_april_tag_detect_detect_args_robot_pose">
      <Field Name="position" DataType="ns=2;i=3" />
      <Field Name="orientation" DataType="ns=2;i=5" />
    </Definition>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
      <Reference ReferenceType="HasEncoding">ns=1;i=8</Reference>
    </References>
  </UADataType>
```
After:
```xml
  <UADataType NodeId="ns=1;i=7" BrowseName="1:rc_april_tag_detect_detect_args_robot_pose">
    <DisplayName>rc_april_tag_detect_detect_args_robot_pose</DisplayName>
    <Description>rc_april_tag_detect_detect_args_robot_pose</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
      <Reference ReferenceType="HasEncoding">ns=1;i=8</Reference>
    </References>
    <Definition Name="rc_april_tag_detect_detect_args_robot_pose">
      <Field Name="position" DataType="ns=2;i=3" />
      <Field Name="orientation" DataType="ns=2;i=5" />
    </Definition>
  </UADataType>
```

Not sure if this is a siome only problem, but it works.

Another point is in the last UAVariable in your file:

```xml
  <UAVariable NodeId="ns=1;i=14" BrowseName="0:InputArguments" ParentNodeId="ns=1;i=13" ValueRank="0" DataType="Argument">
    <DisplayName>InputArguments</DisplayName>
    <Description>InputArguments</Description>
    <References>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=13</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
    </References>
    <Value>
      <uax:ListOfExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=296</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>args_detect</uax:Name>
              <uax:DataType>
                <uax:Identifier>ns=2;i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions />
              <uax:Description>
                 <uax:Encoding>0</uax:Encoding> <!-- there -->
                 <uax:Locale />
                <uax:Text />
              </uax:Description>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
      </uax:ListOfExtensionObject>
    </Value>
  </UAVariable>
```

Not sure what/why this is causing trouble atm, but this is a problem for siome as well...

AndreyKl4, 2021-03-08:
Hi @swamper123,

thank u for your very quick answer! I have done an example that modelate this error and i sent u per Mail! I hope my Code will help u to find an error.

What about Quick-fix. I have done the same already but in my case it is not a good idea to do all fixes manually because i have  a great number of Nodes and XML should be big...

oroulet, 2021-03-08:
I know nothing about xml. how to I evaluate xml output aggainst xsd? changing the order of references and definiton is easy. We/I just need to get a list of things to change.

swamper123, 2021-03-09:
What we would need, is a file, which contains all kind of Nodes to be sure, that everything works as expected. 
I already asked for a Feature Wish in mantis (bug tracker of the foundation), if the Foundation could provide such a file, but haven't received an answer on that.

I checked some other UADAtaTypes in other repos and they all got the references above.

AndreyKl4, 2021-03-10:
@swamper123 Hi, as i understand everything right u need a xsd files. U can find them above in XML file. 

xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">

Or u mean smth else?

oroulet, 2021-03-15:
@AndreyKl4 does it work using master now?

swamper123, 2021-03-16:
@AndreyKl4 I talked about the [UA Nodeset](https://github.com/OPCFoundation/UA-Nodeset), so the "basic" ones from the OPC Foundation.

The longer I read and try stuff, the more difficult the topic in sum becomes (there are even more problems around).

AndreyKl4, 2021-03-17:
@oroulet  sorry for the long answer. I have tested with master - there are no more errors with sequences but another Error in SiOME occur : 

`Validation for C:\Users\klimov\Desktop\test\test_node.xml failed: Line 175, position 18: Das Element 'Description' in Namespace 'http://opcfoundation.org/UA/2008/02/Types.xsd' hat ein ungültiges untergeordnetes Element 'Encoding' in Namespace 'http://opcfoundation.org/UA/2008/02/Types.xsd'. Erwartet wurde die Liste der möglichen Elemente: 'Locale, Text' in Namespace 'http://opcfoundation.org/UA/2008/02/Types.xsd'.`

I found out one more error with namespaces in XML-File. Namespaces in XML-File in some cases not equal with ns in server. That is why some Structures from XML-FIle could not be built in a right way in SiOME or Siemens client for example.

For example ns of node in XML-File:

[image]

ns of node in UaExpert:

[image]

If u want I can send u for tests my server per mail. Also i attach my XML-File.

[test_node.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/6155105/test_node.txt)

Thanks in Advance!

Andrii