FreeOpcUa/opcua-asyncio #488: Attributes of DataTypes are not correctly handled by an opcua server

thunderbug1, 2021-03-18:
I did tests with custom data structures with this [example](https://github.com/FreeOpcUa/opcua-asyncio/blob/3b30f4332e842407a3afbc67ab3af9d64cb401b3/examples/server-create-custom-structures.py) but when I open the server with the [opcua-client ](https://github.com/FreeOpcUa/opcua-client-gui) and navigate to  Root/Types/DataTypes/BaseDataType/Structure then it does not show the attributes of any of the types. Shouldn't there be a detailed definition of what is contained? I also double checked the behaviour with an opcua server on an S7 PLC and there the opcua-client does show attributes for the types, so it must be an issue either with the example or the library.

[image]

swamper123, 2021-03-18:
Have you tried another Client, like UaExpert to verify that this is a server problem?

thunderbug1, 2021-03-18:
So far I only tried the opcua-client-gui, not sure if I can try out UaExpert since I am on a linux machine here.

I was surprised that the gui didn't show any attributes for any of the structure types when connected to the example server.
To check if this is a gui issue I connected to another example server running on the S7 PLC and there the gui does show attributes for all of the structure types, so I don't think it is a client issue.

thunderbug1, 2021-03-18:
oh, I just saw that UaExpert is also available for linux, I will try it out as well.

thunderbug1, 2021-03-18:
Ok I tested it with the UaExpert now, here two screenshots.
Not sure if anything is wrong here but the PLC server sets a lot more attributes as far as I can see.

[image]

[image]

thunderbug1, 2021-04-12:
Does anybody have an idea what causes this problem?

AndreasHeine, 2021-04-12:
how does the req/res look like in wireshark is the result empty or is something in it maybe wrong encoded?

thunderbug1, 2021-04-14:
I had a closer look with wireshark as you suggested and I think it is an encoding issue because all requests are successful but the UaExpert still shows an "Unexpected structure" error.

here a screenshot of the UaExpert
[image]

and the relevant network packets:
[captured_filtered.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/6309055/captured_filtered.zip)

I'm not quite sure at what point the encoding has an issue, the OPC-UA standard is quite bloated and hard to read in that regard.

thunderbug1, 2021-04-14:
I did another test with an open6241 test server to see how the types are displayed there, here a screenshot:

[image]

There are no custom data types here but also for all predefined ones the DataTypeDefinition is "BadAttributeIdInvalid" .. so maybe it is just a very minor bug where instead of this error a freeopcua server gives an unexpected structure.