FreeOpcUa/opcua-asyncio #507: Client connect fails when Client is created outside of asyncio context

Randelung, 2021-03-25:
Creating the `Client` instance outside of an `asyncio` context will lead to a 
```
RuntimeError: Task <Task pending name='Task-2' coro=<BaseEventLoop.create_connection() running at /usr/lib/python3.9/asyncio/base_events.py:1041> cb=[_release_waiter(<Future pendi...3f4873be0>()]>)() at /usr/lib/python3.9/asyncio/tasks.py:416]> got Future <Future pending cb=[BaseSelectorEventLoop._sock_write_done(9, handle=<Handle BaseS...1.10', 4840))>)()]> attached to a different loop
```

The reason is the implicit creation of an asyncio loop in the `Client` initializer.
``` python
import asyncio
import asyncua

client = asyncua.Client('opc.tcp://192.168.1.10:4840/')

async def main():
  await client.connect() # RuntimeError
  while True:
    asyncio.sleep(1)

if __name__ == '__main__':
  asyncio.run(main())
```

The initializer should not implicitly create the loop, as it's making assumptions about the calling scope. (If no loop is available but one is necessary, throw an exception, but it appears no tasks are scheduled at initialization time, so it shouldn't be required.)
The same applies to `Server`. A `Server` instance needs an `init()` call, anyway, and a client would need to be `connect()`ed first, which are better positions to grab the running event loop, as they're both `async`.

A quick fix would be to use [`get_running_loop`](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_running_loop) instead of `get_event_loop` (which will conflict with python 3.6 and just throw an exception), but a nicer solution would be to initialize `self.loop` at a later point inside an async method or alternatively make `self.loop` a property to get the running loop the first time it's needed.

The issue is obviously easily worked around when initializing inside the first `async` block.

``` python
import asyncio
import asyncua

client = None

async def main():
  global client
  client = asyncua.Client('opc.tcp://192.168.1.10:4840/')
  await client.connect() # works fine
  while True:
    asyncio.sleep(1)

if __name__ == '__main__':
  asyncio.run(main())
```

Python-Version: 3.9.2
opcua-asyncio Version (e.g. master branch, 0.9): pip 0.9.14

AndreasHeine, 2021-03-25:
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/README.md

removing support of python < 3.7

oroulet, 2021-03-25:
you might be into something here. Earlier we had to keep track of loop object all the time, but as far as I understand the best practice now is to use the current loop.
We should remove all the self.loop and passing loop everywhere. It should work, although I have not looked at code in detail for that particular case

oroulet, 2021-03-26:
I started something there: https://github.com/FreeOpcUa/opcua-asyncio/pull/510

oroulet, 2021-07-19:
The Pr over is merged so I close. reopen if any issue left

closed 2021-07-19 by oroulet