FreeOpcUa/opcua-asyncio #509: load_data_type_definitions failling, worked in 0.9.0 (load_type_definitions)

jcbastosportela, 2021-03-25:
Client connection to OPC UA server on a B&R PLC with version 0.9.0 was working fine with `load_type_definitions`. Trying to use the PR: #508 that delivers some functionality I need.

When running with deprecated `load_type_definitions` I get the following error:
```
INFO:asyncua.client.ua_client.UaClient:activate_session
WARNING:asyncua.client.client:Deprecated since spec 1.04, call load_data_type_definitions
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 32, in <module>
    asyncio.run(main())
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 19, in main
    await client.load_type_definitions()
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\client\client.py", line 619, in load_type_definitions
    return await load_type_definitions(self, nodes)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures.py", line 222, in load_type_definitions
    generator.get_python_classes(structs_dict)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures.py", line 176, in get_python_classes
    return _generate_python_class(self.model, env=env)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures.py", line 279, in _generate_python_class
    exec(code, env)
  File "<string>", line 3, in <module>
  File "<string>", line 12, in FetchResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
```

When running with the recommended `load_data_type_definitions` I get the following error: 
```
INFO:asyncua.client.ua_client.UaClient:activate_session
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) TraceLevel
INFO:asyncua.common.structures104:Registring data type NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=1, Name='TraceLevel')
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 32, in <module>
    asyncio.run(main())
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 19, in main
    await client.load_data_type_definitions()
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\client\client.py", line 627, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 305, in load_data_type_definitions
    new_objects = await load_enums(server)  # we need all enums to generate structure code
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 375, in load_enums
    env = await _generate_object(name, edef, enum=True)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 238, in _generate_object
    exec(code, env)
  File "<string>", line 16
SyntaxError: can't assign to keyword
```

**Version**<br />
Python-Version:3.7 and 3.8<br />
opcua-asyncio Version: master

Thank you for any help on this regards.
Regards,
Portela

oroulet, 2021-03-26:
yes there are many dangerous changes in master. but very nice if you can help me debug that.
Can you update master and run again? and send the output here. I added a print in case of error. The load_data_type_definition( )should be easy to fix at least. you probably have a struct with a strange name that conflicts

jcbastosportela, 2021-03-26:
I checked out your PR's last commit that had the changes for printing the "code" and the try catch.

This is what I get:

```
INFO:asyncua.client.ua_client.UaClient:activate_session
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) TraceLevel
INFO:asyncua.common.structures104:Registring data type NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=1, Name='TraceLevel')
ERROR:asyncua.common.structures104:Failed to execute auto-generated code from UA datatype: 

class TraceLevel(IntEnum):

    '''
    TraceLevel EnumInt autogenerated from EnumDefinition
    '''

    Error = 32
    Warning = 48
    System = 56
    Info = 60
    Debug = 62
    Content = 63
    All = -1
    None = 0
Traceback (most recent call last):
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 239, in _generate_object
    exec(code, env)
  File "<string>", line 16
SyntaxError: can't assign to keyword
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 34, in <module>
    asyncio.run(main())
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 19, in main
    await client.load_data_type_definitions()
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\client\client.py", line 627, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 309, in load_data_type_definitions
    new_objects = await load_enums(server)  # we need all enums to generate structure code
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 379, in load_enums
    env = await _generate_object(name, edef, enum=True)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 239, in _generate_object
    exec(code, env)
  File "<string>", line 16
SyntaxError: can't assign to keyword
```

Thanks you for the super fast replies,
Portela

oroulet, 2021-03-26:
can you update and try again?

jcbastosportela, 2021-03-26:
> can you update and try again?

So, I checked out da8be3767fda6a2c70f7beed70fd3aae61312a4c
The result is:

```
INFO:asyncua.client.ua_client.UaClient:activate_session
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) TraceLevel
INFO:asyncua.common.structures104:Registring data type NodeId(Identifier=3002, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=1, Name='TraceLevel')
ERROR:asyncua.common.structures104:Failed to execute auto-generated code from UA datatype:

class TraceLevel(IntEnum):

    '''
    TraceLevel EnumInt autogenerated from EnumDefinition
    '''

    Error = 32
    Warning = 48
    System = 56
    Info = 60
    Debug = 62
    Content = 63
    All = -1
    None = 0
Traceback (most recent call last):
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 241, in _generate_object
    exec(code, env)
  File "<string>", line 16
SyntaxError: can't assign to keyword
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 26, in <module>
    asyncio.run(main())
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Program Files (x86)\Python37-32\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "c:/BnR/exe_ftplc/Tests/regressionTests/client-data-type-definition.py", line 16, in main
    await client.load_data_type_definitions()
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\client\client.py", line 627, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 311, in load_data_type_definitions
    new_objects = await load_enums(server)  # we need all enums to generate structure code
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 381, in load_enums
    env = await _generate_object(name, edef, enum=True)
  File "c:\BnR\exe_ftplc\Tests\regressionTests\asyncua\common\structures104.py", line 241, in _generate_object
    exec(code, env)
  File "<string>", line 16
SyntaxError: can't assign to keyword
```

oroulet, 2021-03-26:
try again ;-)

jcbastosportela, 2021-03-28:
> try again ;-)

Just tried, it is now working.

closed 2021-03-28 by jcbastosportela