FreeOpcUa/opcua-asyncio #520: #461 causing connection errors

swamper123, 2021-03-30:
**Describe the bug** <br />
Since #461 Connection errors happen in UaExpert.

**To Reproduce**<br />
Run server-minimal.py and connect via UaExpert

**Expected behavior**<br />
No troubles should be caused while connecting.

**Screenshots**<br />
[image]

**Version**<br />
Python-Version: 3.8<br />
opcua-asyncio Version (e.g. master branch, 0.9): master

#461 should be reverted and fixed.

oroulet, 2021-03-30:
Can you revert then? doing that was a bit of an issue anyway

swamper123, 2021-03-30:
Couldn't push it on master on my own, so we have to do it via PR.

Koseng, 2021-03-30:
Hi,
I basically just took that PR https://github.com/FreeOpcUa/python-opcua/pull/1197, which was recommended here: https://github.com/FreeOpcUa/python-opcua/issues/850 and did not do any further tests.

So we should check what is causing the trouble. I'll have a look.

closed 2021-04-03 by AndreasHeine