FreeOpcUa/opcua-asyncio #522: test_cli_tools has runtimeerror issue

swamper123, 2021-03-30:
**Describe the bug** <br />
test_cli_tools causes a Runtime error -> sometimes tests fail.

Log from a failed test:

```
 ________________________________ test_cli_tools ________________________________

running_server = 'opc.tcp://127.0.0.1:40399'

    async def test_cli_tools(running_server):
        # admin privileges are only needed for uawrite
        url = running_server.replace("//", "//admin@")
        default_opts = ["mock_func", "-u", f"{url}"]
        rw_node = ["-n", RW_NODE]
        call_node = ["-n", ROOT_NODE]
        write_val = ["val_to_write"]
        rw_opts = default_opts + rw_node
        write_opts = rw_opts + write_val
        call_opts = default_opts + call_node + write_val
    
        tool_opts = {}
        tool_opts[uaread] = rw_opts
        tool_opts[uals] = rw_opts
        tool_opts[uawrite] = write_opts
        tool_opts[uahistoryread] = rw_opts
        tool_opts[uaclient] = default_opts
        tool_opts[uadiscover] = default_opts
        tool_opts[uacall] = call_opts
    
        for tool in (uaread, uals, uawrite, uahistoryread, uaclient, uadiscover, uacall):
            # It's necessary to mock argv, else the tool is invoked with *pytest's* argv
            with patch.object(sys, 'argv', tool_opts[tool]):
                with concurrent.futures.ThreadPoolExecutor(max_workers=1) as executor:
                    result = executor.submit(tool)
                    for future in concurrent.futures.as_completed([result]):
                        exception = future.exception()
                        # py3.6 returns SystemExit(0,)
                        str_exp = repr(exception).replace(",", "")
>                       assert str_exp == "SystemExit(0)"
E                       assert 'ConnectionRe....1\' 40399)")' == 'SystemExit(0)'
E                         - SystemExit(0)
E                         + ConnectionRefusedError(111 "Connect call failed ('127.0.0.1' 40399)")

tests/test_tools.py:45: AssertionError
------------------------------ Captured log setup ------------------------------
INFO     root:internal_server.py:65 No user manager specified. Using default permissive manager instead.
INFO     asyncua.server.internal_session:internal_session.py:45 Created internal session Internal
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=11715, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15958, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15959, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15960, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15961, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15962, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15963, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=15964, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=16134, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=16135, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncua.server.address_space:address_space.py:245 add_node: while adding node NumericNodeId(Identifier=16136, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO     asyncio:base_events.py:1748 poll 2977.607 ms took 2986.267 ms: timeout
------------------------------ Captured log call -------------------------------
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=85, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=85, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=85, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=7, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=7, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=7, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=7, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>))
INFO     asyncua.common.instantiate_util:instantiate_util.py:98 Instantiate: Skip node without modelling rule QualifiedName(NamespaceIndex=2, Name='ServerMethodDefault') as part of QualifiedName(NamespaceIndex=2, Name='ObjectWithMethods')
INFO     asyncua.common.instantiate_util:instantiate_util.py:98 Instantiate: Skip node without modelling rule QualifiedName(NamespaceIndex=2, Name='ServerMethodNone') as part of QualifiedName(NamespaceIndex=2, Name='ObjectWithMethods')
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=85, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>))
INFO     asyncua.common.manage_nodes:manage_nodes.py:161 create_method Node(NodeId(Identifier=85, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>))
INFO     asyncua.common.xmlimporter:xmlimporter.py:93 Importing XML file /home/runner/work/opcua-asyncio/opcua-asyncio/tests/enum_struct_test_nodes.xml
INFO     asyncua.client.client:client.py:238 connect
INFO     asyncua.client.ua_client.UaClient:ua_client.py:264 opening connection
```

**To Reproduce**<br />
Difficult, since this happens not every time a test is pushed.
If you rerun a failed test, it may pass afterwards.

**Expected behavior**<br />
Test should pass everytime, unless something is realy broken.

swamper123, 2021-04-14:
Do you @ZuZuD and  @oroulet (you made the last changes here) have a clue, where this can be caused? Since there is a comment about python3.6, is this still "state of the art"?

ZuZuD, 2021-04-14:
> Do you @ZuZuD and @oroulet (you made the last changes here) have a clue, where this can be caused? Since there is a comment about python3.6, is this still "state of the art"?

Py3.6 was returning a different exception but since we deprecated py3.6 this hack is not required anymore.

There's arguably a race when the server opens its socket and the client tries to connect, ECONNREFUSED is returned when there's no port listening:
```
> strace -f -e getsockopt nc -vz 127.0.0.1 1201
Ncat: Version 7.70 ( https://nmap.org/ncat )
getsockopt(3, SOL_SOCKET, SO_ERROR, [ECONNREFUSED], [4]) = 0
Ncat: Connection refused.
```

The logs you pasted @swamper123 confirm we didn't reach [srv.start()](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/tests/conftest.py#L75-L86) yet. Since it's a separate thread, the best workaround is probably to retry the client call. Let me see what I can do in that regard.

swamper123, 2021-04-26:
Case closed with #550  - 🕵️

closed 2021-04-26 by swamper123