FreeOpcUa/opcua-asyncio #536: BadWriteNotSupported error

sdfzz, 2021-04-06:
Hi all,

I'm trying to write an OPCUA client to control an OPCUA server (UA-7560M from ICP DAS), and got "BadWriteNotSupported: the server does not support writing the combination of value..." error.

Searching the forum here, I've found that this issue have been discussed a few times...but the proposed solutions did not work on my case:

dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
dv.ServerTimestamp = datetime.now()
dv.SourceTimestamp = datetime.now()

or 

dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
dv.ServerTimestamp = None
dv.SourceTimestamp = None

However, the OPCUA server works with the opcua-client, and checking the opcua-client, I've found that the server requires SourceTimestamp but not ServerTimestamp.

[image]

So, I've tried:

dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
dv.ServerTimestamp = None
dv.SourceTimestamp = datetime.now()

and got the same BadWriteNotSupported error...

One thing I've noticed from the opcua-client is that the timestamp format differs from what I've got by datetime.now():
opcua-client   : 2021-04-05T10:38:47:911768
datatime.now(): 2021-04-05 17:38:47:911768 (see the pic above)

Not sure where the letter "T" comes from, but I guess it is the cause of the error here. Does anyone know how to fix the problem, or how I can get the same timestamp as the opcua-client? 

Any help would be appreciated

Sincerely

Steve

AndreasHeine, 2021-04-06:
why do you pass a list with just one item? if its scalar just pass "True"!

> use datetime.utcnow() 

all times are utc!!!

sdfzz, 2021-04-06:
Thank you for the quick reply

Unfortunately, changing list nor using utcnow() solved the problem....still getting BadWriteNowSupported error...

AndreasHeine, 2021-04-06:
which version of asyncua?

AndreasHeine, 2021-04-06:
and please provide the full stacktrace!

sdfzz, 2021-04-06:
Thank you for the response. I'm sorry but I've just left the lab....I will post update tomorrow.

Also, would you please tell me how I can check my asyncua version? I simply installed it with pip command.

AndreasHeine, 2021-04-06:
https://pip.pypa.io/en/stable/reference/pip_list/

sdfzz, 2021-04-08:
I'm so sorry, but the device is working now, although I didn't make any changes

Must have been a mistake in my code....I can control variables using:

dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
dv.ServerTimestamp = None
dv.SourceTimestamp = None

I would like to close the issue. 

Thank you so much for your help

closed 2021-04-08 by sdfzz