FreeOpcUa/opcua-asyncio #545: what is the correct way to change the StatusCode of a variable?

mar-ar, 2021-04-08:
Hey guys,

I want to change the StatusCode of a variable and receive an callback if I have subscribed to this node.

`        
dv = await node.read_attribute(ua.AttributeIds.Value)
dv.StatusCode = StatusCode(StatusCodes.BadDataUnavailable)
dv.SourceTimestamp = datetime.utcnow()
await node.write_attribute(ua.AttributeIds.Value, dv)
`

I did that above, but then I dont get the callback.
Does soemone know a way that everything works fine?
I took a look in the code, but I couldnt find an mistake.

Thanks for your help

AndreasHeine, 2021-04-08:
you should:

```python
import asyncio
from asyncua import Server, ua

async def main():
    server = Server()
    await server.init()
    node = await server.get_objects_node().add_variable(0, "0:Test", 0, ua.VariantType.Int64)
    async with server:
        while 1:
            await asyncio.sleep(2)
            dv = ua.DataValue(
                ua.Variant(5, ua.VariantType.Int64),
                ua.StatusCode(ua.StatusCodes.Good)
            )
            await server.write_attribute_value(node.nodeid, dv)
            await asyncio.sleep(2)
            dv = ua.DataValue(
                ua.Variant(5, ua.VariantType.Int64),
                ua.StatusCode(ua.StatusCodes.BadDataUnavailable)
            )
            await server.write_attribute_value(node.nodeid, dv)

asyncio.run(main())

```

tested with UaExpert!

mar-ar, 2021-04-08:
Thanks for your help! This works fine for me, but the other way should also work.
I did some further debuging. Both ways  results in the same function in address_space.write_attribute_value but there the old StatusCode already is like the new one.

But your solution works fine thanks!

closed 2021-04-08 by mar-ar

oroulet, 2021-04-08:
the syntax in your example over will not work on master. DataValue are now read-only

AndreasHeine, 2021-04-08:
@oroulet means you need to recreate the DataValue-Instance if you want to change timestamps?

oroulet, 2021-04-09:
@AndreasHeine yes. so your example is correct, but mar-ar example will not work

AndreasHeine, 2021-04-09:
ok thanks! i need to check some of my code now 🙃

oroulet, 2021-04-09:
Also ideally we could set all ua structure frozen. That removed completely the risk for typos and make memory allocation faster since we only allocate once instead of two. But there is too much legacy code for such a thing so I just changes the most used structures for now...
Maybe in asyncua v2...

AndreasHeine, 2021-04-09:
what i am thinking about is that i cant change the timestamp after creating the DataValue or we need to recreate a new DataValue from the old one in order to set a missing timesteamp

oroulet, 2021-04-09:
yes that one is an issue. Maybe I should revert the change for DataValue just because of that one.. Or maybe it is completely fine to rcreate a new DataValue. Most probably Python is only moving some pointer addresse and not allocating new memory

AndreasHeine, 2021-04-09:
@oroulet 

```python
                    try:
                        dv = await parse_to_datavalue(item)
                    except Exception as e:
                        print(e)
                        dv = None

                    if dv is not None:
                        new_dv = ua.DataValue(
                            Value=dv.Value,
                            StatusCode_=dv.StatusCode_,
                            SourceTimestamp=dv.SourceTimestamp,
                            ServerTimestamp=datetime.utcnow()
                        )
                        await server.write_attribute_value(nodeid, new_dv, ua.AttributeIds.Value)
```

if i make something like this in a loop (current project writing csv data to nodes) it will trigger always a datachangenotification even if the statuscode and value are the same ...

oroulet, 2021-04-09:
because you are changing the itmestamp?

AndreasHeine, 2021-04-09:
https://github.com/FreeOpcUa/opcua-asyncio/blob/3b974f1407fa4c90d14a13ae8b18e20e22aef8e1/asyncua/server/address_space.py#L713

it should not... did i messed up!?

oroulet, 2021-04-09:
I have no ideas. you put a breakpoint there and see the difference

AndreasHeine, 2021-04-09:
yeah i did ^^ it changed to Null and back to the Value but in UaExpert you only see the last change it happens to fast (-_-)° sorry!