FreeOpcUa/opcua-asyncio #546: Reading Node using Ua.NodeId errors in asyncio

nagamanikandank, 2021-04-09:
The following code works fine

```
import asyncio
from opcua import ua
from opcua import Client

client = Client("opc.tcp://127.0.0.1:49311")
client.connect()
one_node = client.get_node(ua.NodeId("Simulator.Test.Node1", 2))
node_value = one_node.get_value()
print(node_value)
```

But, the async code below errors with the message ```asyncua.ua.uaerrors._base.UaError: argument to node must be a NodeId object or a string defining a nodeid found StringNodeId(ns=2;s=Simulator.Test.Node1) of type <class 'opcua.ua.uatypes.NodeId'>```

I am accessing the node data the same way I am accessing without async, but it errors

```
import asyncio
import time
from opcua import ua

from asyncua import Client

class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("New data change event", node, val)

    def event_notification(self, event):
        print("New event", event)

async def main():
    url = "opc.tcp://127.0.0.1:49311"
    async with Client(url=url) as client:
        while True:
            time.sleep(30)
            one_node = client.get_node(ua.NodeId("Simulator.Test.Node1", 2))
            node_value = one_node.get_value()
            print(node_value)

if __name__ == "__main__":
    asyncio.run(main())

```

AndreasHeine, 2021-04-09:
ua.NodeId.from_string("ns=2;s=Simulator.Test.Node1")

oroulet, 2021-04-09:
ua.NodeId("Simulator.Test.Node1", 2) also works perfectly. I do dnot see where the issue is

closed 2021-07-19 by oroulet