FreeOpcUa/opcua-asyncio #548: BUG in Reference DisplayName in Master broken!

AndreasHeine, 2021-04-14:
If you look in the UaExpert ReferenceTab an browse a Node it does not say HasTypeDefinition instead it says HasDescription!?
ReferenceTypeId is ok only the DisplayName looks broken.

v0.9.14 is ok

i would assume those commits broke it somewhere:
https://github.com/FreeOpcUa/opcua-asyncio/commit/6bdfbbaf14c5932949af4e1d5ab105fae9231383
https://github.com/FreeOpcUa/opcua-asyncio/commit/ebf5d40c81b36ad121552f1cb650b0d22d0d4965

oroulet, 2021-04-14:
Prosys has the issue, why the us client has not... strange... what do they do difrently. 
and the reference definition in code is correct, with correct browse name...

oroulet, 2021-04-14:
fixed there: https://github.com/FreeOpcUa/opcua-asyncio/pull/549
It was a bug introduced while parsing the xml file instead of deprecated xd

oroulet, 2021-04-14:
thanks for the reporting

AndreasHeine, 2021-04-14:
tbh https://github.com/ccvca reported to me that the server shows weird things in the references ^^

AndreasHeine, 2021-04-14:
@oroulet 
a release in the near future would be nice and something was strange in the last 0.9.14 should be 13 i guess -> there was something strange the commit says 13 and the release has 14 ^^

case closed thanks!

closed 2021-04-14 by AndreasHeine

oroulet, 2021-04-14:
There are so many fundamental changes in master that I am afraid to release it. for example you just found a big bug today....
Also we had a bug parsing custom structs at work last time we tried on one of the real robots. I need to try again