FreeOpcUa/opcua-asyncio #555: Running example code in VSC debug mode is very slow on startup

g3n35i5, 2021-04-30:
**Describe the bug** <br />
When I run the example `example/sync/server-minimal.py` in DEBUG mode of VSC, the initialization of the server takes extremely long. In comparison: If I execute the code via terminal, it is much faster. For comparison I measured the time the code needs before the `server.start()` method is executed:

Debug mode: 21.151774 seconds
Command Line: 1.124574 seconds

This is my `launch.json` settings file:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal"
        }
    ]
}
```

**To Reproduce**<br />
Run example minimal server code both, in VSC debug mode and from command line

**Expected behavior**<br />
Of course the execution in debug mode may be a bit slower... But not so much slower :D

**Version**<br />
Python-Version: 3.7.9<br />
opcua-asyncio: Version 0.9.14

oroulet, 2021-04-30:
there is one HUGE file defining the opcua address space. I am guessing that is the one giving issues to VCS. This is a limitation of VCS we cannot do much about it

g3n35i5, 2021-04-30:
Thanks @oroulet  for your answer! Which file do you mean exactly? Could this be a memory issue?

oroulet, 2021-04-30:
might be... standard_address_space_service.py I think the file is called

g3n35i5, 2021-05-01:
Thanks for your help! I will experiment a little bit in the next days but I close the issue for now. If I find out something else that can help, I'll report back here.

closed 2021-05-01 by g3n35i5

stark-alexander, 2022-12-19:
@oroulet Is there any way to reduce the size of this huge file or to speed up the initialization? With the intel pentium 4x1.1GHz CPU, the initialization takes about 10 seconds. But using the open62541 library with reduced address spaces < 100ms.

oroulet, 2022-12-19:
One could implement some reduce address space. Many strange stuff there. PR welcome.
Otherwise it is possible to cache that file. I think there examples for that