FreeOpcUa/opcua-asyncio #561: Error while renewing session. AttributeError: 'NoneType' object has no attribute 'write'

Ilyazyk, 2021-05-04:
I use a sync version of client with Rockwell FactoryTalk Linx Gateway Server.
Security mode is None.
I have a subscriprion for around 1000 tags, and sometimes write values. After some time (usually less than 1 hour, but not sure) I got the error.

Can it be a case: when client doesn't write for a long time a session closes by timeout?
I will be keeping monitoring system logs and last write time before error.

 
**Error logged:** <br />
ERROR:asyncua.client.client:Error while renewing session
Traceback (most recent call last):
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python39\lib\site-packages\asyncua\client\client.py", line 399, in _renew_channel_loop
    await self.open_secure_channel(renew=True)
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python39\lib\site-packages\asyncua\client\client.py", line 292, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python39\lib\site-packages\asyncua\client\ua_client.py", line 278, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python39\lib\site-packages\asyncua\client\ua_client.py", line 213, in open_secure_channel
    await asyncio.wait_for(self._send_request(request, message_type=ua.MessageType.SecureOpen), self.timeout)
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python39\lib\site-packages\asyncua\client\ua_client.py", line 134, in _send_request
    self.transport.write(msg)
AttributeError: 'NoneType' object has no attribute 'write'

**Version**<br />
Python-Version: 3.9.4
opcua-asyncio Version: 0.9.14 + most recent master version
Rockwell FactoryTalk Linx Gateway Server version: 6.20.02.
OS: Windows 2016

swamper123, 2021-05-04:
Yes, renewing a session is not handled properly at the moment and is on the TODO list.
The workaround atm is to catch the error, disconnect in a clean way (unsub etc.), reconnect and subscribe again.
Maybe you can gain some inspiration from [this example](https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/39beee1c466d66cb00742e37298b1b11ddf24053/async_client.py#L68).

Ilyazyk, 2021-05-04:
Update:
In 10 minutes after last write the client closed connection (which I read in system logs of server), but I realised that on new write attempt and got an error:(<class 'ConnectionError'>, ConnectionError('Connection is closed'). 
It seems 10 minutes is expected timeout setting as I got on client start a warning: "asyncua.client.client:Requested session timeout to be 3600000ms, got 600000ms instead".
How this can be controlled? Obviosly, I don't want session to be closed by itself. The option can be a handling an event of connection closing and reopen it. Is it possible?

oroulet, 2021-05-04:
OK. that sounds a bit strange, What is your secure channel timeout?

We have a task supposed to read a node withing the spesified timeout.
check the methods get_keepalive_count() and _renew_channel_loop(), add some debug code there and see of they are called when they should

Also if you have no subscription and a secure channel timeout which is longer than session timeout you may have encountered that one: https://github.com/FreeOpcUa/opcua-asyncio/pull/439

@ZuZuD if you donnot want  to renew secure channel too often, then we might need a new task just for the session

ZuZuD, 2021-05-04:
I see 2 potential use-cases for @Ilyazyk issue:

1. There's indeed no subscription (but @Ilyazyk said otherwise in the initial message), so the user has to make sure the `secure_channel_timeout` is inferior to the `session_timeout` (as specified in #339).

2. There's not much activity with the subscription (only receives empty PublishResponses) and the KeepAlive should kick in to keep the session active. However, since the KA is based on the `RequestedPublishingInterval`, it can differ from the actual server `RevisedPublishingInterval` and KA might start too late if it has a lower value, see #557.

Let's see what comes out of the debug logs.

AndreasHeine, 2021-05-04:
What if MaxKeepAliveCount is reached?

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.1/
Subscriptions have a lifetime counter that counts the number of consecutive publishing cycles in which there have been no Publish requests available to send a Publish response for the Subscription. Any Service call that uses the SubscriptionId or the processing of a Publish response resets the lifetime counter of this Subscription. When this counter reaches the value calculated for the lifetime of a Subscription based on the MaxKeepAliveCount parameter in the CreateSubscription Service (5.13.2), the Subscription is closed. Closing the Subscription causes its MonitoredItems to be deleted. In addition the Server shall issue a StatusChangeNotification notificationMessage with the status code Bad_Timeout. The StatusChangeNotification notificationMessage type is defined in 7.20.4.

AndreasHeine, 2021-05-04:
a wireshark trace would be nice to see what happens!?

Ilyazyk, 2021-05-04:
There are some settings on subscription creation time, if it helps:
session_timeout = 600 000, 
get_keepalive_count(period=0.25) -> 1 800 000, 
secure_channel_timeout': 3 600 000
RequestedPublishingInterval = 0.25

Writing that I've discovered an inconsistency with units of measure: 0,25 (Sec) vs 600 000 (mSec). Looks like a root cause for me. Could you confirm? If yes, may be some warning can be added in case of suspicious (foolish) settings.

@ZuZuD, the subscription exits and works at least at begining. And I can expect no activity on both: subscription and write values as nothing changes. Will it help, if periodically write some node's value (even if it's not changed)?
@AndreasHeine, if it's still needed, could you share any ideas of what to monitor by wireshark (a config is welcome)?

Ilyazyk, 2021-05-04:
@oroulet 
with my 'wrong' setpoint of period = 0.25 mSec:
get_keepalive_count(period=0.25) -> 1 800 000 (is called twice)
_renew_channel_loop() is not called at all in a time range: since start till connection close (+ several minutes after).

AndreasHeine, 2021-05-04:
you can trace and later filter for opcua protocol and export only the marked lines (like last working renew till crash)

Ilyazyk, 2021-05-04:
I keep testing with a period = 0.25 mSec.
asyncua debug.log is attached ([debug.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/6421590/debug.zip)).

16:42:37 - started, a lot of nodes were written, subscription is initialized
16:45:20 - one node write initiated (my action)
16:55:21 - connection was closed (not my action)
16:56:54 - application was stopped (my action)

So, 10 minutes timeout were counted since last write.

Is "auto-reniew session" option supposed to be in place I should take care of to don't exceed timeout and write some node?

@AndreasHeine, I could not manage opcua protocol to be logged, by just stat trace. Should it be somehow activated?

ZuZuD, 2021-05-04:
> I see 2 potential use-cases for @Ilyazyk issue:
>
>     1. There's indeed no subscription (but @Ilyazyk said otherwise in the initial message), so the user has to make sure the `secure_channel_timeout` [...]

You hit the use-case 2). 

What's happening with this low PublishInterval value is that the server likely doesn't support it and uses a higher value instead. The KeepAlive is still expected to trigger after a count of 1 800 000 PublishInternval (like initially calculated by the client), but since the actual value is higher, the session times out before the server sends it.

@Ilyazyk short-term fix: set the `RequestedPublishingInterval` to at least the minimal value supported by your server (check the documentation or test with a value of 0 and log the `PublishResponse` timestamp).
Long-term generic fix: implement #557

Ilyazyk, 2021-05-04:
Good that you guys know how to fix that. Looking forward to see it in reliase.
So far, I've applied another short-term fix: I've added an item "ua.ObjectIds.Server_ServerStatus_CurrentTime" into subscription (as in example, btw), which keeps it busy. Thanks for support!

closed 2021-05-04 by Ilyazyk