FreeOpcUa/opcua-asyncio #564: Register to discovery not working

SeiteBroken, 2021-05-06:
Im new to opcua-asyncio and i tried this code

```
import logging
import asyncio
import sys
sys.path.insert(0, "..")

from asyncua import ua, Server
from asyncua.common.methods import uamethod

@uamethod
def func(parent, value):
    return value * 2

async def main():
    _logger = logging.getLogger('asyncua')

    server = Server()
    await server.init()
    server.set_endpoint('opc.tcp://0.0.0.0:4841/freeopcua/server/')

    uri = 'http://examples.freeopcua.github.io'
    idx = await server.register_namespace(uri)
    

    await server.register_to_discovery("opc.tcp://localhost:4840", 60)

    myobj = await server.nodes.objects.add_object(idx, 'MyObject')
    myvar = await myobj.add_variable(idx, 'MyVariable', 6.7)

    await myvar.set_writable()
    await server.nodes.objects.add_method(ua.NodeId('ServerMethod', 2), ua.QualifiedName('ServerMethod', 2), func, [ua.VariantType.Int64], [ua.VariantType.Int64])
    _logger.info('Starting server!')
    async with server:
        while True:
            await asyncio.sleep(1)
            new_val = await myvar.get_value() + 0.1
            _logger.info('Set value of %s to %.1f', myvar, new_val)
            await myvar.write_value(new_val)

if __name__ == '__main__':

    logging.basicConfig(level=logging.DEBUG)

    asyncio.run(main(), debug=True)

````
I only got this from the LDS log:

````
Thu May 06 15:51:39 [5644]: ualds_endpoint_callback called: Event=SecureChannelOpened, SecureChanneldId=0x00000002, uStatus=0x00000000
Thu May 06 15:51:39 [5644]: ualds_endpoint_callback called: Event=SecureChannelOpenVerifyCertificate, SecureChanneldId=0x00000002, uStatus=0x800B0000
````

And this from the terminal:

````
DEBUG:asyncio:Using proactor: IocpProactor
INFO:root:No user manager specified. Using default permissive manager instead.
INFO:asyncua.server.internal_session:Created internal session Internal
INFO:asyncua.server.address_space:add_node: while adding node i=15957, requested parent node i=11715 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15958, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15959, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15960, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15961, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15962, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15963, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=15964, requested parent node i=15957 does not exists
INFO:asyncua.server.address_space:add_node: while adding node i=16134, requested parent node i=15957 does not exists
[... 32 lines cut ...]
WARNING:asyncua.client.ua_client.UaClient:disconnect_socket was called but connection is closed
DEBUG:asyncio:Close <ProactorEventLoop running=False closed=False debug=True>
Traceback (most recent call last):
  File "c:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\server.py", line 49, in <module>
    asyncio.run(main(), debug=True)
  File "C:\Users\benni\AppData\Local\Programs\Python\Python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\benni\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "c:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\server.py", line 28, in main
    await server.register_to_discovery("opc.tcp://localhost:4840", 60)
  File "C:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\.venv\lib\site-packages\asyncua\server\server.py", line 239, in register_to_discovery
    await self._discovery_clients[url].connect()
  File "C:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\.venv\lib\site-packages\asyncua\client\client.py", line 244, in connect
    await self.create_session()
  File "C:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\.venv\lib\site-packages\asyncua\client\client.py", line 366, in create_session
    response = await self.uaclient.create_session(params)
  File "C:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\.venv\lib\site-packages\asyncua\client\ua_client.py", line 299, in create_session
    data = await self.protocol.send_request(request)
  File "C:\Users\benni\OneDrive\Dokumente\Neuer Ordner\Python\.venv\lib\site-packages\asyncua\client\ua_client.py", line 150, in send_request
    raise ConnectionError("Connection is closed") from None
ConnectionError: Connection is closed
````

Can somebody tell me where my mistake is ?

AndreasHeine, 2021-05-06:
your discoveryserver is not a discoveryserver -> MessageAbort(error:StatusCode(BadServiceUnsupported)

AndreasHeine, 2021-05-06:
how is the discoveryserver configured?

SeiteBroken, 2021-05-06:
I used this LDS Release:
https://opcfoundation.org/developer-tools/samples-and-tools-unified-architecture/local-discovery-server-lds/

AndreasHeine, 2021-05-06:
ok! 
looks like we have a bug -> https://reference.opcfoundation.org/v104/GDS/docs/4.2.2/

we only call register and not register2 up front...

https://github.com/FreeOpcUa/opcua-asyncio/blob/af5a31c0db2b4f2469cdc7c169a0aa812b606152/asyncua/server/server.py#L227

SeiteBroken, 2021-05-06:
Okay, so there is a bug in asycua.
Thank you!

AndreasHeine, 2021-05-06:
i can reproduce the error but the weird part is i cant connect to the lds at all it always throws BadServiceUnsupported...

AndreasHeine, 2021-05-08:
https://reference.opcfoundation.org/v104/Core/docs/Part7/5.2/
ServerDiscovery   
RegisterCall
 the RegisterServer Service to register itself (OPC UA Server) with an external Discovery Service via a secure channel with a SecurityMode other than NONE.

i am not sure that we need a session for that!?

https://github.com/FreeOpcUa/opcua-asyncio/blob/af5a31c0db2b4f2469cdc7c169a0aa812b606152/asyncua/server/server.py#L227

https://github.com/FreeOpcUa/opcua-asyncio/blob/af5a31c0db2b4f2469cdc7c169a0aa812b606152/asyncua/client/client.py#L233

AndreasHeine, 2023-02-05:
@SeiteBroken https://github.com/FreeOpcUa/opcua-asyncio/pull/1193