FreeOpcUa/opcua-asyncio #571: Create "writeable" variables

untixx, 2021-05-17:
I'm using asyncua in combination with open62541 server which doesn't allow the `node.set_writable(True)` after client node creation. E.g. the following does **not** work:

```python
async with Client(url="opc.tcp://user1:password@localhost:4840/") as client:
  objects = client.nodes.objects
  o = await objects.add_object(1, 'MyObject')
  v = await o.add_variable(1, 'MyVariable', 23, varianttype=ua.VariantType.Int32)
  await v.set_writable(True)
  await v.write_value(5, varianttype=ua.VariantType.Int32)
```

It raises `asyncua.ua.uaerrors._auto.BadUserAccessDenied: "User does not have permission to perform the requested operation."(BadUserAccessDenied)` on `v.set_writable(True)`. By default (no call of `v.set_writable(True)`) it raises `asyncua.ua.uaerrors._auto.BadNotWritable: "The access level does not allow writing to the Node."(BadNotWritable)`.

Getting out of this misery requires setting the nodes access level on node creation. My fix would be to add "writeable" parameter to `add_variable`, `create_variable` and `_create_variable` and adding these two lines to `manage_nodes.py`:

```python  
if writeable:
  attrs.AccessLevel |= ua.AccessLevel.CurrentWrite.mask
  attrs.UserAccessLevel |= ua.AccessLevel.CurrentWrite.mask
```

Then the following code does work:

```python
async with Client(url="opc.tcp://user1:password@localhost:4840/") as client:
  objects = client.nodes.objects
  o = await objects.add_object(1, 'MyObject')
  v = await o.add_variable(1, 'MyVariable', 23, varianttype=ua.VariantType.Int32, writeable=False)
  await v.write_value(5, varianttype=ua.VariantType.Int32)
```

Is it possible to integrate this (or something similar) into asyncua, or am I understanding some OPC UA concept completely wrong?

oroulet, 2021-05-17:
not sure why they let you create a node, then do not let you change its attribute. Are you sure it is not a bug on their side?
Otherwise the simple solution is to send a AddNodeItems request directly. look at code of create_node and copy what you need

swamper123, 2021-05-17:
It may sound trivial, but has your user the needed permission rights, like the error mentioned?

untixx, 2021-05-18:
> not sure why they let you create a node, then do not let you change its attribute. Are you sure it is not a bug on their side?

Definitely not sure, as I have no idea how it *should* behave. The access rights for node management are quite confusing when looking at the open62541 implementation. Even in the reference documents they don't care about [AccessLevelTypes](https://reference.opcfoundation.org/Core/docs/Part3/8.57/) for Variable NodeClasses ([CurrentRead -> OPC_READABLE CurrentWrite -> OPC_WRITABLE The other AccessLevel provided by OPC are ignored](https://reference.opcfoundation.org/Core/docs/Part8/A.4.2/) - whatever this means?!). Why are there so many specifications and they did not specify the default behavior of new nodes or something like AccessLevelTypes for node management? I'am not getting warm with OPC UA.

> Otherwise the simple solution is to send a AddNodeItems request directly. look at code of create_node and copy what you need

Thanks, I just copied the relevant functions and created a subclass of Node. For reference my implementation:

```python

from asyncua import Client, ua, Node
from asyncua.common import manage_nodes, node_factory

async def _create_variable(server, parentnodeid, nodeid, qname, var, datatype=None, isproperty=False, writeable=False):
    addnode = ua.AddNodesItem()
    addnode.RequestedNewNodeId = nodeid
    addnode.BrowseName = qname
    addnode.NodeClass = ua.NodeClass.Variable
    addnode.ParentNodeId = parentnodeid
    if isproperty:
        addnode.ReferenceTypeId = ua.NodeId(ua.ObjectIds.HasProperty)
        addnode.TypeDefinition = ua.NodeId(ua.ObjectIds.PropertyType)
    else:
        addnode.ReferenceTypeId = ua.NodeId(ua.ObjectIds.HasComponent)
        addnode.TypeDefinition = ua.NodeId(ua.ObjectIds.BaseDataVariableType)
    attrs = ua.VariableAttributes()
    attrs.Description = ua.LocalizedText(qname.Name)
    attrs.DisplayName = ua.LocalizedText(qname.Name)
    if datatype:
        attrs.DataType = datatype
    else:
        attrs.DataType = manage_nodes._guess_datatype(var)

    attrs.Value = var
    if not isinstance(var.Value, (list, tuple)):
        attrs.ValueRank = ua.ValueRank.Scalar
        attrs.ArrayDimensions = None
    else:
        if var.Dimensions:
            attrs.ValueRank = len(var.Dimensions)
            attrs.ArrayDimensions = var.Dimensions
    attrs.WriteMask = 0
    attrs.UserWriteMask = 0
    attrs.Historizing = False

    attrs.AccessLevel = ua.AccessLevel.CurrentRead.mask
    attrs.UserAccessLevel = ua.AccessLevel.CurrentRead.mask
    if writeable:
        attrs.AccessLevel |= ua.AccessLevel.CurrentWrite.mask
        attrs.UserAccessLevel |= ua.AccessLevel.CurrentWrite.mask
    addnode.NodeAttributes = attrs
    results = await server.add_nodes([addnode])
    results[0].StatusCode.check()
    return results[0].AddedNodeId

async def create_variable(parent, nodeid, bname, val, varianttype=None, datatype=None, writeable=False):
    """
    create a child node variable
    args are nodeid, browsename, value, [variant type], [data type]
    or idx, name, value, [variant type], [data type]
    """
    nodeid, qname = manage_nodes._parse_nodeid_qname(nodeid, bname)
    var = ua.Variant(val, varianttype)
    if datatype and isinstance(datatype, int):
        datatype = ua.NodeId(datatype, 0)
    if datatype and not isinstance(datatype, ua.NodeId):
        raise RuntimeError("datatype argument must be a nodeid or an int refering to a nodeid")

    return node_factory.make_node(
        parent.server,
        await _create_variable(parent.server, parent.nodeid, nodeid, qname, var, datatype=datatype, isproperty=False,
                               writeable=writeable)
    )

class WriteableNode(Node):
    def __init__(self, server, nodeid):
        super(WriteableNode, self).__init__(server, nodeid)

    async def add_variable(self, nodeid, bname, val, varianttype=None, datatype=None, writeable=False):
        return await create_variable(self, nodeid, bname, val, varianttype, datatype, writeable)

async def main():
    async with Client(url="opc.tcp://user1:password@localhost:4840/") as client:
        objects = client.nodes.objects
        o = await objects.add_object(1, 'MyObject2')
        wo = WriteableNode(o.server, o)
        v = await wo.add_variable(1, 'MyVariable', 6, varianttype=ua.VariantType.Int32, writeable=True)
        await v.write_value(2, varianttype=ua.VariantType.Int32)
```

Not elegant but working.

oroulet, 2021-05-18:
you could have mad it much simpler by making a function instead of creating so many classes but yes that is fine to do it that way

closed 2021-05-18 by untixx