FreeOpcUa/opcua-asyncio #573: Subscribe to OPC UA events (CODESYS)

mozesa, 2021-05-19:
**Describe the bug** <br />
I cannot subscribe to OPC UA events.
I think the problem comes from `get_filter_from_event_type`, more precisely `evfilter.SelectClauses` elements' `TypeDefinitionId` is  `NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>)`.

It should be `NodeId(Identifier=2782, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>)`.

**To Reproduce**<br />
```python
import asyncio

from asyncua import Client
from asyncua.ua import CreateSubscriptionParameters

class SubscriptionHandler:
    def event_notification(self, event):
        print(f"{event=}")

async def main():
    client = Client(url="opc.tcp://127.0.0.1:4840/")
    async with client:
        obj = await client.nodes.root.get_child(
            [
                "0:Objects",
                "2:DeviceSet",
                "4:CODESYS Control Win V3 x64",
                "4:Application",
            ]
        )
        events = await client.nodes.root.get_child(
            [
                "0:Types",
                "0:EventTypes",
                "0:BaseEventType",
                "0:ConditionType",
            ]
        )

        # We create a Client Subscription.
        handler = SubscriptionHandler()

        params = CreateSubscriptionParameters()
        params.RequestedPublishingInterval = 500
        params.RequestedLifetimeCount = 2400
        params.RequestedMaxKeepAliveCount = 10
        params.MaxNotificationsPerPublish = 10_000
        params.PublishingEnabled = 1
        params.Priority = 0

        sub = await client.create_subscription(params, handler)
        handle = await sub.subscribe_events(obj, events)
        print("*** Sleep ***")
        await asyncio.sleep(60)
        await sub.unsubscribe(handle)
        await sub.delete()

if __name__ == "__main__":
    asyncio.run(main())
```

**Expected behavior**<br />
I would like to see the events.

**Screenshots**<br />
[image]
[image]

**Version**<br />
Python-Version: 3.9.5<br />
opcua-asyncio Version (e.g. master branch, 0.9): master branch

mozesa, 2021-05-21:
The problem comes from the not proper `get_filter_from_event_type` function.
The event type is `i=2782`, but the function creates a filter for `i=0`.

[image]

mozesa, 2021-05-26:
Am I right or I miss something?
Currently I use a monkey patch to overcome this issue.

How can I help you solving this issue?

Thanks for your help in advance.

mozesa, 2021-05-28:
I created an overview - you can reach it [here](https://github.com/FreeOpcUa/opcua-asyncio/discussions/572#discussioncomment-796555)

oroulet, 2021-07-19:
Can you make a PR of your patch? Looks like you known better than everyone here what the correct behaviour is. I do not use events currently so I do not remember much

mozesa, 2021-07-19:
Unfortunately I couldn't patch it properly.
It would require more thorough examination of the event's implementation itself - the problem is buried there.

What I have described is still valid.

(Currently I am coding in Golang so no Python on the horizon.)

closed 2023-03-29 by schroeder-