FreeOpcUa/opcua-asyncio #586: Method timeouterror

sdfzz, 2021-06-03:
Python 3.8
Ubuntu 18.04
asyncua 0.9.10

Hi,

I'm trying to use opcua-asyncio for my robotic project. I setup my robot as an OPCUA server so that my client can control it via methods. Once the task given by a method is complete, I want it to return True/False to indicate if the task was successful or not. 

However, I've found that if the task inside a method takes more than 3 seconds, I get timeout error (error code given below). I've tested it by calling the following method:

server: 
```
@uamethod
def test(parent):
    delay = 3
    time.sleep(delay)
    return true
```
client: 
`res = await obj.call_method("0:test")`

My project requires tasks with long duration. Is there anyway to increase timeout?

Sincerely,

Steve

error code:
```
    res = await obj.call_method("0:test")
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/common/node.py", line 697, in call_method
    return await call_method(self, methodid, *args)
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/common/methods.py", line 19, in call_method
    result = await call_method_full(parent, methodid, *args)
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/common/methods.py", line 43, in call_method_full
    result = await _call_method(parent.server, parent.nodeid, methodid, to_variant(*args))
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/common/methods.py", line 63, in _call_method
    results = await server.call(methodstocall)
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 612, in call
    data = await self.protocol.send_request(request)
  File "/home/zenbook/python3/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 147, in send_request
    data = await asyncio.wait_for(self._send_request(request, timeout, message_type), timeout if timeout else None)
  File "/usr/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

AndreasHeine, 2021-06-03:
https://github.com/FreeOpcUa/opcua-asyncio/blob/9acce70431d20706c5bdc6704bb543f6571f7a4a/asyncua/client/client.py#L32

try a bigger timeout value in client class cunstructor!

sdfzz, 2021-06-03:
Dear AndreasHeine,

Thank you so much for your answer. Problem solved.

Steve

closed 2021-06-03 by sdfzz

oroulet, 2021-06-03:
Also generally long network calls are a bad idea. They will fail one day. It is much more reliable to have one call to start a job and then another call to poll for result.

AndreasHeine, 2021-06-03:
was that the subtle hint that I should finally finish the missing statemachines?
its one of its usecases^^ "#programstatemachine"

AndreasHeine, 2021-06-03:
@sdfzz you dont have to return anything except maybe a bad status code if the call fails like "BadNotExecutable" retuning just a boolean is kind of ... in opc ua because you can use the statuscodes for that.

by the way the opc robotics group has a companionspec. you should check it out
https://reference.opcfoundation.org/Robotics/docs/

sdfzz, 2021-06-03:
oroulet, AndreasHeine, 

Thank you for the tips. Guess I better start changing my state machine to avoid long calls.

Again, thank you for all your help

Steve

prrw, 2023-08-18:
@oroulet @AndreasHeine I am implementing it in a state machine but i am getting this error inside the run(). Basically i am running a for loop to copy lots of values from sensor to OPC UA which is taking more than 4 seconds and it results in asyncio Timeout exception. After overriding the client with a higher timeout value it works. But now i am thinking what is that is causing this problem in a state machine. The overall run() loop goes on for 12 seconds but only the for loop if taking more than 4 seconds is raising the exception. If the length of array in the for loop is small then Timeout is not raised. Please find the snippet below:

```python
    async def run(self):
        logging.info('Started executing XYZ skill')
        await asyncio.sleep(0.5)

        # Does something here (altogether takes 8 seconds or so)
        # Upto here there is no exception raised

                for row in xyz:                                    # [ During
                    point = ua.ThreeDCartesianCoordinates()        # these lines
                    point.X = row[0]/1000.0                        # the Timeout
                    point.Y = row[1]/1000.0                        # exception
                    point.Z = row[2]/1000.0                        # is
                    points.append(point)                           # raised ]
                await self.outputs.PointCloud.write_value(points)) # This line should be fine

            # If everything is fine go back to the ready state 
            await self.change_state(self.ready, event_msg='XYZ skill finished')
            logging.info('Finished executing XYZ skill')
            
    async def onEntryRunning(self):
        logging.info('Entering running state for XYZ skill')
        asyncio.create_task(self.run())

```
On the server side there is no error. Everything is smooth and the method succeeds as soon as it is called and the FinalResultData is written later when the run() is executed. What is the problem during the for loop here. Is it advisable to use for loops for async operations ?

AndreasHeine, 2023-08-18:
the error usually happens when there is a request which does not get an response in the 4s timeout range. 

your client gets the error? not the server?

AndreasHeine, 2023-08-18:
async iterators and generators could be used but it depends on the used libraries...

prrw, 2023-08-18:
> the error usually happens when there is a request which does not get an response in the 4s timeout range.
> 
> your client gets the error? not the server?

only the client gets error. Server continues. If i comment out the for loop, it works although the entire duration of the run() is still 8 seconds in that case.

prrw, 2023-08-18:
> what does the client do? read? call? write?

The client writes the inputs parameters of the skill, executes the start method of the skill and later reads the output (FinalResultData). Timeout error is raised in the client while executing the start method (which triggers the run() in the server)

AndreasHeine, 2023-08-18:
ah ok then the method does not return and you get no repsonse on clientside results in error....

AndreasHeine, 2023-08-18:
you need to create a task and return from the uamethod rather then waiting for the inner stuff to finish

prrw, 2023-08-18:
write, call and read all three in the client are used with await in front. 

> you need to create a task and return from the uamethod rather then waiting for the inner stuff to finish

Thanks. Any examples for this in the repository ?

AndreasHeine, 2023-08-18:
not in the repo but this is from one of my projects (nothing to do with opc ua!)

https://docs.python.org/3/library/asyncio-task.html#creating-tasks
```python
  self.__inc_request_count()
  task = create_task(
      self._process_request(item)
  )
  task.add_done_callback(
      self.__dec_request_count
  )

```

prrw, 2023-08-18:
Okay. Thanks a lot. 👍  I can figure it out now.

AndreasHeine, 2023-08-18:
you should build something around to keep track of running and pending tasks to be safe!