FreeOpcUa/opcua-asyncio #611: Scrambled DataType NodeIds when using multiple information models

jcbastosportela, 2021-06-23:
**Describe the bug** <br />
When importing multiple information models, where DataTypes are defined across them and used in the topmost information model, the server will report incorrect Namespace Index for the DataTypes of Varaibles (for example).

As far as I could investigate the issue is in `common\xmlimporter.py::_to_migrated_nodeid` which is then calling `common\xmlimporter.py::_migrate_ns` for nodeids which are `aliases`. AFAI understood the alias are already in the server namespace index, therefore `common\xmlimporter.py::_migrate_ns` should not be called.

I have made a fix that resolves my problem, but I am not sure if it is the correct approach for this:

```python
   def _to_migrated_nodeid(self, nodeid: Union[ua.NodeId, None, str]) -> ua.NodeId:
        needs_migration, nodeid = self._to_nodeid(nodeid)
        if needs_migration: # added this check for migrating namespace conditionally
            nodeid = self._migrate_ns(nodeid)
        return nodeid

    def _to_nodeid(self, nodeid: Union[ua.NodeId, None, str]) -> bool, ua.NodeId:  # returns a pair of values, bool indicates need of ns migration
        if isinstance(nodeid, ua.NodeId):
            return True, nodeid
        if not nodeid:
            return True, ua.NodeId(ua.ObjectIds.String)
        if "=" in nodeid:
            return True, ua.NodeId.from_string(nodeid)
        if hasattr(ua.ObjectIds, nodeid):
            return True, ua.NodeId(getattr(ua.ObjectIds, nodeid))
        if nodeid in self.aliases:
            return False, self.aliases[nodeid]  # aliases do not need migration
        return True, ua.NodeId(getattr(ua.ObjectIds, nodeid))
```

**To Reproduce**<br />
Create 3 information models, each of them defining one custom DataType.
Create other information model which uses the 3 previously created information models; in this one create an ObjectType that contains one instance on each of the DataTypes from the other information models.

You may use the attached files [multiple_inf_models.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/6701602/multiple_inf_models.zip) __ATTENTION__: in order for this information models to work it is necessary to apply the fix I suggest in https://github.com/FreeOpcUa/opcua-asyncio/issues/594#issuecomment-866763055 (Because all of the information models generated by current UaModeler version defines TypeDictionary with same BrowseName Name, but different BrowseName NamespaceIndex)

**Expected behavior**<br />
In the client the correct DataType for each of the variables is reported.

**Screenshots**<br />
This is what I get:
[image]

This is what I expect to get:
[image]

**Version**<br />
Python-Version:3.7 and 3.8 (at least, most likely all)
opcua-asyncio Version (e.g. master branch, 0.9): tested in 0.8.2 and in master.

jcbastosportela, 2021-06-29:
@oroulet is there a way I can help with resolving this issue? I could make a PR with the changes I mention, but I honestly don't know if that is the real solution; it does resolve my problem, but it may create other problems.

oroulet, 2021-06-29:
I do not have time to  look at xml import, we are not using that at work. Also I do not see what you changed, please make a diff or better a merge request and we can see what you propose to change.
I do not know if alias should be migrated, probably not since they are all in namespace 0

swamper123, 2021-06-29:
The aliases have to be migrated, since you don't know the order of required models in the xml path (so you have to check and combine the ns right again). If the Alias is not a standard one and has it's source in an XML-Model, the ns has to be migrated. 
Example: [MachineTool CS](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/MachineTool/Opc.Ua.MachineTool.NodeSet2.xml)   `  <Alias Alias="StacklightOperationMode">ns=4;i=3002</Alias>`

But it seems to be an importer thing, indeed, but I haven't got an adhoc solution for that atm.

jcbastosportela, 2021-06-29:
> The aliases have to be migrated, since you don't know the order of required models in the xml path (so you have to check and combine the ns right again). If the Alias is not a standard one and has [...]

Indeed! This is exactly my case.

PR #615 (2021-07-06): Fix #611 scrambled datatypes nodeids
fixes #611 
also https://github.com/FreeOpcUa/opcua-asyncio/issues/594#issuecomment-866763055

__NOTE__ : Sorry for the previous PR. It was having changes from my fork's master where I applied https://github.com/FreeOpcUa/opcua-asyncio/pull/517#issue-602313624 because I need it. In this PR I remove those changes (sorry for the trailing of unrelated commits you see with this PR, I should not have changed my fork's master :( )
commit f30430ab (2021-07-06): fixed double namespace translation of aliases. Fixes #611 (#615)
closed 2021-07-06 by oroulet