FreeOpcUa/opcua-asyncio #621: UnicodeDecodeError When Reading Node Value

sanssecours, 2021-07-15:
# Description

The library seems to have problem when reading string values that contain incorrect UFT-8 byte sequences. It would be nice, if there would be an option to replace these byte sequences with a replacement character such as `�`.

# Steps to Reproduce the Problem

1. Paste the following Python code

   ```py
   from asyncio import run
   from pathlib import Path
 
   from asyncua import Client
   from asyncua.ua import NodeId
 
 
   async def read_value():
       cert = Path(__file__).parent.joinpath("cert.der")
       key = Path(__file__).parent.joinpath("key.pem")
       uri = "opc.tcp://SuperCoolUsername:SuperSafePassword@localhost:4840"
       security = f"Basic256Sha256,SignAndEncrypt,{cert},{key}"
       client = Client(uri)
       await client.set_security_string(security)
 
       async with client as client:
           return (await client.read_values([
               client.get_node(NodeId("/Channel/ProgramInfo/actPartProgram", 2))
           ])).pop()
 
 
   if __name__ == '__main__':
       print(run(read_value()))
   ```

2. Replace the values `SuperCoolUsername` and `SuperSafePassword` with actual values that work for your OPC UA server
3. Save the text as `error.py`
4. Generate the key `key.pem` and certificate `cert.der` with a slightly modified version of [`generate_certificate.sh`](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/generate_certificate.sh) and put them into the same folder as the script.
5. Execute the Python script

   ```sh
   python error.py
   ```

# Expected Result

The script prints the value of the node `/Channel/ProgramInfo/actPartProgram`:

```

MCHOME
TLCH1H("M3016FR",-90) ;Werkzeugwechsel h�ndisch

TLCOSOF
MCMILLS1
G54
D1
G95 S3=294 M3=3 F=0.45  ;M3=8
;MCICOTON

G0 X1=160 Y1=0 C1=150
G0 Z1=-11
G1 X1=-20
G0 X1=0
G1 Y1=25
```

The value seems to contain an Umlaut character (`ä`) (in ISO-8859-1 encoding). The screenshot below shows the value in the Prosys OPC UA Browser:

[image]

Here is the first part of the text that contains the problematic byte sequence at the end:

```py
b'\nMCHOME\r\nTLCH1H("M3016FR",-90) ;Werkzeugwechsel h\xe4'
```

The problematic value `\xe4` represents the Umlaut `ä` character in the ISO-8859-1 encoding:

```py
>>> b'\xe4'.decode('iso-8859-1')
'ä'
```

# Actual Result

The program fails with an `UnicodeDecodeError`:

```
Traceback (most recent call last):
  File "/Users/rene/Documents/Work/Projects/AUDIo/CNC Connect/error.py", line 23, in <module>
    print(run(read_value()))
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/rene/Documents/Work/Projects/AUDIo/CNC Connect/error.py", line 17, in read_value
    return await client.read_values([
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/client/client.py", line 666, in read_values
    results = await self.uaclient.read_attributes(nodeids, ua.AttributeIds.Value)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/client/ua_client.py", line 668, in read_attributes
    response = struct_from_binary(ua.ReadResponse, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 492, in struct_from_binary
    val = from_binary(uatype, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 464, in from_binary
    return unpack_uatype_array(vtype, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 226, in unpack_uatype_array
    return [unpack_uatype(vtype, data) for _ in range(length)]
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 226, in <listcomp>
    return [unpack_uatype(vtype, data) for _ in range(length)]
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 199, in unpack_uatype
    return struct_from_binary(cls, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 492, in struct_from_binary
    val = from_binary(uatype, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 469, in from_binary
    return unpack_uatype(vtype, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 195, in unpack_uatype
    return variant_from_binary(data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 379, in variant_from_binary
    value = unpack_uatype(vtype, data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 187, in unpack_uatype
    return st.unpack(data)
  File "/Users/rene/.pyenv/versions/3.9.6/lib/python3.9/site-packages/asyncua/ua/ua_binary.py", line 72, in unpack
    return b.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 49: invalid continuation byte
```

# Environment

Python: `Python 3.9.6`
opcua-asyncio: `0.9.14`

# Other Information

A very similar issue was already reported in [`python-opcua`](https://github.com/FreeOpcUa/python-opcua/issues/1222).

oroulet, 2021-07-19:
thanks for the bug report. That clear. I implemented what you requested: https://github.com/FreeOpcUa/opcua-asyncio/pull/632
Can you try and let me know if it works?

sanssecours, 2021-07-20:
>  I implemented what you suggested: #632

Thanks for the quick update.

> Can you try and let me know if it works?

As far as I can tell, everything works as expected now.

closed 2021-07-20 by sanssecours