FreeOpcUa/opcua-asyncio #633: v0.9.14 -> v0.9.90 breaks NodeId interface

joristork, 2021-07-20:
**Description** <br />
The upgrade from v0.9.14 to v0.9.90 breaks the interface of NodeId.

**To Reproduce**<br />
Steps to reproduce the behavior incl code:
Write code that initialises a `NodeId` from `asyncua==0.9.14` as follows:
```python
NodeId(identifier=2259, namespaceidx=0, nodeidtype=None)
``` 
Test to assert this code works.
Upgrade to `asyncua==0.9.90`.

**Expected behavior**<br />
The code should continue to work, as `0.9.14` to `0.9.90` is a [patch version upgrade](https://semver.org/).

**Version**<br />
Python-Version: 3.7.6<br />
opcua-asyncio Version (e.g. master branch, 0.9): 0.9.90

oroulet, 2021-07-20:
Shit i knew i was forgetting something when releasing yesterday.
Next major number is 1 so i choose 0.90 instead, you are welcome to propose a better idea.
But i should really have written a big warning about that breaking change.
The constructors are now autogenerated and use the UA cases: Identifier

oroulet, 2021-07-20:
Also FYI i added a few wrappers to remove the worst things like argument order inversions. But some renaming generating clear errors and making things much more  consistent were left

joristork, 2021-07-21:
In my CI, I rely on the [semver](https://semver.org/) convention, wherever it seems to be respected, to allow automatic patch upgrades but not minor version upgrades (e.g. for your library, `asyncua >=0.9.14, <0.10.0`). You could make the breaking version `0.10.0` at least, even though it doesn't comply with semver to have breaking changes in a minor update... Any reason that you don't want to bump to `1.0.0` ? I guess it's a roadmap issue?

oroulet, 2021-07-21:
I do not want 1.0 so people do not think this is a stable product yet. It is not like i am paid to maintain it. I just happen to need some functionalities

joristork, 2021-07-23:
Meh, stability, what is that? :)
semver is your friend when it comes to communicating the instability, though

closed 2023-03-29 by schroeder-