FreeOpcUa/opcua-asyncio #642: Struct typedefinition is wrong

SArzbach, 2021-08-09:
I'm running a OPC-UA server on an Omron NX102-9000.
Im using structs to combine variables.
The library doesn't return the Typeinformation and the ID of the type node is off by one.
When im using load_type_definitions() i'm getting an Exception.
get_value() is running, but returns a wrong Datatype

```python
print(await systeminfo.get_value())
```

the dataFEED client can read the Typeinformation just fine:
[image]

get value returns the correct content, but the Type is wrong and the type ID is the correct type ID plus one
[image]

```python
print(await self.client.client.load_type_definitions([systeminfo]))
```

The load_type_definitions function runs into an exception:
[image]

Versions:
Python-Version: 3.9
opcua-asyncio Version: 0.9.90

AndreasHeine, 2021-08-09:
have you tried load_data_type_definitions ?

https://github.com/FreeOpcUa/opcua-asyncio/blob/085b00e41e98a33230f735d25597007a8b5f099d/asyncua/client/client.py#L664

SArzbach, 2021-08-09:
> 
> 
> have you tried load_data_type_definitions ?
> 
> https://github.com/FreeOpcUa/opcua-asyncio/blob/085b00e41e98a33230f735d25597007a8b5f099d/asyncua/client/client.py#L664

now i have 
it just returns an empty dictionary without a specified node as well as with the node in question as variable
the return of the get_value call also remains the same

oroulet, 2021-08-09:
I am impressed how few servers support the 1.04 specification.... you should check with omron if the ua server on PLc can be updated. 1.04 really clean up the custom struct implementation. (load_data_type_definition() uses the new protocol, load_type_definition the old one, which imports an xml in the background)

oroulet, 2021-08-09:
if there is an offsett, then someone introduced a bug in xml importer in that case. when importing xml from server, we should not migrate the namespaces, they are already correct

SArzbach, 2021-08-09:
Update: 
```python
systeminfo.read_data_type()
```
Does return the correct TypeId, NodeId

SArzbach, 2021-08-10:
@oroulet the highest opc-ua version possible would be 1.03

AndreasHeine, 2021-08-10:
it would be interesting if the issue is also in python-opcua!?
https://github.com/FreeOpcUa/python-opcua

oroulet, 2021-08-10:
obs sorry I did not read carefully the bug description. looks like you are passing the wrong node to load_type_definition(). The value of the nodes should be xml text, you are obvisouly passing a node having some extension object. What happens when you run load_type_definition() wihtout argument. Doesn't it work?

SArzbach, 2021-08-10:
> obs sorry I did not read carefully the bug description. looks like you are passing the wrong node to load_type_definition(). The value of the nodes should be xml text, you are obvisouly passing a [...]

Thank you that solved it.

closed 2021-08-10 by SArzbach