FreeOpcUa/opcua-asyncio #651: Exception in xmlimporter while loading the MachineVision nodeset

honkomonk, 2021-08-20:
**Describe the bug** <br />
I tried to import the MachineVision companion specs [xml file](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/MachineVision/Opc.Ua.MachineVision.NodeSet2.xml), but just importing it raised an exception in the importer (see below). Looks like an unexpected object type was passed internally.

```
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=2002, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>))

Traceback (most recent call last):
  ...
  File " ... /opcua-asyncio/examples/server-robotics.py", line 12, in main
    nodes1 = await server.import_xml("../schemas/UA-Nodeset-master/MachineVision/Opc.Ua.MachineVision.NodeSet2.xml")
  File "..\asyncua\server\server.py", line 561, in import_xml
    return await importer.import_xml(path, xmlstring)
  File "..\asyncua\common\xmlimporter.py", line 108, in import_xml
    node = await self._add_node_data(nodedata, no_namespace_migration=True)
  File "..\asyncua\common\xmlimporter.py", line 188, in _add_node_data
    node = await self.add_variable_type(nodedata, no_namespace_migration)
  File "..\asyncua\common\xmlimporter.py", line 457, in add_variable_type
    if obj.value and len(obj.value) == 1:
TypeError: object of type 'ExtObj' has no len()
```

**To Reproduce**<br />
I took the `server-robotics.py` script and just dropped in the path to the machine vision nodeset:
```
import asyncio
import logging
import sys
sys.path.insert(0, "..")
from asyncua import Server

async def main():
    logging.basicConfig(level=logging.INFO)
    server = Server()
    await server.init()
    # import some nodes from xml
    nodes1 = await server.import_xml("../schemas/UA-Nodeset-master/MachineVision/Opc.Ua.MachineVision.NodeSet2.xml")
    # nodes1 = await server.import_xml("../schemas/UA-Nodeset-master/DI/Opc.Ua.Di.NodeSet2.xml")
    # nodes2 = await server.import_xml("../schemas/UA-Nodeset-master/Robotics/Opc.Ua.Robotics.NodeSet2.xml")

    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())
```

**Version**<br />
Python-Version:  3.8<br />
opcua-asyncio Version (e.g. master branch, 0.9): just pulled the latest version of the master branch (20.08.2021)

jblereste, 2022-02-16:
Hello,

I still have the same bug: commit 033427b6a5bdf1a4a512e4976a410b6cc6291406 (origin/master)

How can I help you to debug it ? 

Thanks,
Jean-Baptiste

neixlo, 2022-04-19:
Same here.

@honkomonk or @jblereste could one of you resolve the issue?

neixlo, 2022-05-09:
Thanks @honkomonk for sharing your findings!

If possible, could you share your modified xml file?

schroeder-, 2023-03-29:
#1259 should fix this

closed 2023-03-31 by schroeder-