FreeOpcUa/opcua-asyncio #655: Creating a server with OPC 40077 fails, due to importer errors

GoetzGoerisch, 2021-09-01:
**Describe the bug** <br />
While importing the nodeset [40083 GeneralTypes](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PlasticsRubber/GeneralTypes/1.03/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml) the importer fails with:

```
Importing companion spec. XML...
('Error no extension class registered ', 'PIDParametersDataType', NodeId(Identifier=3023, NamespaceIndex=7, NodeIdType=<NodeIdType.Numeric: 2>))
```

As per discussion with @AndreasHeine we might need to create structs on the fly, while importing.

**To Reproduce**<br />
Steps to reproduce the behavior incl code:
Implement server as per: https://github.com/umati/Sample-Server-asyncio/pull/40 
with `logging.basicConfig(level=logging.DEBUG)`

**Expected behavior**<br />
The the companion spec is imported and instanciated correctly.

**Screenshots**<br />
The PR above.

**Version**<br />
Python-Version: **3.9**<br />
opcua-asyncio Version (e.g. master branch, 0.9): **0.9.90**

KUZJurk, 2022-01-26:
For me the Error is still present when trying to import on sync or on async server.
I attached a log file to this post for the importing procedure.

__Version__

Python-Version: __3.7.9__

opcua-asyncio Version: __0.9.92_

[GeneralTypes_Errorlog.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/7941129/GeneralTypes_Errorlog.txt)
_

swamper123, 2022-07-25:
This issue is still up-to-date:

```
Exception: ('Error no extension class registered ', 'PIDParametersDataType', NodeId(Identifier=3023, NamespaceIndex=4, NodeIdType=<NodeIdType.Numeric: 2>))
```

This DataType was not added to the `extension_objects_by_typeid` while importing, which .

Do you have any clue what could be the cause  @oroulet ?

schroeder-, 2022-07-25:
@swamper123 maybe this recent patch helps here: #947.

GoetzGoerisch, 2022-07-25:
Thank you @schroeder- 
Tried with the latest commit: https://github.com/umati/Sample-Server-asyncio/pull/40/commits/f8da71ec2e6e823aef4222832aee3724f59422a5

Still no luck.
```
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=6318, NamespaceIndex=8, NodeIdType=<NodeIdType.Numeric: 2>)) maximum recursion depth exceeded while calling a Python object
WARNING:asyncua.common.xmlimporter:Model is missing: http://opcfoundation.org/UA/PlasticsRubber/GeneralTypes/ - Version: 1.03.01 - PublicationDate: 2022-07-13T00:00:00Z or newer
Traceback (most recent call last):
  File "/src/server.py", line 300, in <module>
Start setup...
Setup done! 1.9531893730163574s
Importing companion spec. XML...

UpdateBehavior = ua.UInt32

Handle = ua.UInt32

maximum recursion depth exceeded while calling a Python object
Server doesn't satisfy required XML-Models. Import them first!
    asyncio.run(main())
  File "/usr/local/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/src/server.py", line 137, in main
    pr_imm2mes_idx = await server.get_namespace_index("http://opcfoundation.org/UA/PlasticsRubber/IMM2MES/")
  File "/src/asyncua/asyncua/server/server.py", line 499, in get_namespace_index
    return uries.index(uri)
ValueError: 'http://opcfoundation.org/UA/PlasticsRubber/IMM2MES/' is not in list
```

schroeder-, 2022-07-25:
You have to to add the parameter: `strict_mode=False`
`await server.import_xml(os.path.join(BASE_DIR,  "nodeset", "Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml"), strict_mode=False)`
Then all errors are converted to warnings. Maybe this helps to load the nodeset until we fix the underlying issue.

GoetzGoerisch, 2022-07-25:
Thank you @schroeder- 
It works a bit further, at least it starts to import 40077 but throws a lot of Error due to ExtensionObjects. Still does not start the server.

schroeder-, 2022-08-22:
I checked it and found the following:

1. ExtensionObjects: the xml endcoding is only a stub that works for basic types. So there are some errors that needs to be tracked down.
2. Fields in DataTypes with no DataType pr: #1008
3. DefaultBinary and DefaultXML nodes without _ParentNodeId_ for example ns=1;i=5038. Don't know if this is a error in the nodeset , because every Variable node should have a _ParentNodeId_.

This modified nodeset loads:
[Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/9393182/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.txt)

GoetzGoerisch, 2022-10-17:
Thanks @schroeder-  still don't work for me. See above.
Now the max recursion depth is exceeded.
[e2c1f39](https://github.com/DrHaraldWeber/Sample-Server-asyncio/commit/e2c1f39731dd0475f58f9f83d00da08fdc50ae00)

schroeder-, 2023-03-29:
With #1259 everything works without any changes or strict_mode=False

GoetzGoerisch, 2023-03-30:
Can confirm from my side: https://github.com/umati/Sample-Server-asyncio/commit/b257fb489ecc757c344f994083f4526102a80c50

closed 2023-03-31 by schroeder-