FreeOpcUa/opcua-asyncio #657: Error while renewing session - open_secure_channel -> BadTcpSecureChannelUnknown

vvelghe88, 2021-09-03:
**Describe the bug**
When the asyncua client trigger the _renew_channel_loop (after 75% of secure_channel_timeout), an exception is thrown :
```
Error while renewing session
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/asyncua/client/client.py", line 400, in _renew_channel_loop
    val = await self.nodes.server_state.read_value()
  File "/usr/local/lib/python3.8/dist-packages/asyncua/common/node.py", line 177, in read_value
    result = await self.read_data_value()
  File "/usr/local/lib/python3.8/dist-packages/asyncua/common/node.py", line 188, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value)
  File "/usr/local/lib/python3.8/dist-packages/asyncua/common/node.py", line 302, in read_attribute
    result = await self.server.read(params)
  File "/usr/local/lib/python3.8/dist-packages/asyncua/client/ua_client.py", line 357, in read
    data = await self.protocol.send_request(request)
```

Once the timeout is reach, another exception is thrown : 
```
Received an error: ErrorMessage(Error=StatusCode(value=2155806720), Reason='Could not verify security on incoming request.')
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/asyncua/client/ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "/usr/local/lib/python3.8/dist-packages/asyncua/client/ua_client.py", line 103, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/usr/local/lib/python3.8/dist-packages/asyncua/client/ua_client.py", line 169, in _call_callback
    raise ua.UaError(f"No request found for request id: {request_id}, pending are {self._callbackmap.keys()}, body was {body}") from ex
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([1070]), body was "The SecureChannelId and/or TokenId are not currently in use."(BadTcpSecureChannelUnknown)
Request to close socket received
```

**Timeline (when secure_channel_timeout = 60s)**: 
2021-09-03T12:29:39Z [info     ] Client connected to server 
2021-09-03T12:30:24Z [error   ] Error while renewing session
2021-09-03T12:31:24Z [error   ] Received an error: ErrorMessage(Error=StatusCode(value=2155806720), Reason='Could not verify security on incoming request.')

This last error force a new (re)connection

**To Reproduce**
Start the client and wait for the secure_channel_timeout

**Expected behavior**
The SecureChannel is correctly renewed

**Version**
Python-Version: 3.8
opcua-asyncio Version: 0.9.90

Please, help me on this. I hard tried to fix this without any success :(

FShinoda, 2023-04-06:
I am facing the same issue

schroeder-, 2023-04-06:
Do you have a full trace? Because the error hints that the secure channel was closed or never established correct before _renew_channel_loop is called.

schroeder-, 2023-04-06:
Also would be helpfull to know which server/SDK/toolkit is used. Or if there is a public test server available to test against.