FreeOpcUa/opcua-asyncio #661: Can't write UInt16 value to a property

xmas79, 2021-09-14:
Hello,
It seems I'm not able to write a UInt16 value to a property. There's must be something strange on how the data type is managed internally and I cannot figure out how to write... Below you can find a simple demo server which explains the issue and several attempts. 

Could someone please help me on this? 

```
import asyncio
import opcua.ua as ua

from asyncua import Server

async def main():
    server = Server()
    await server.init()

    server.set_endpoint("opc.tcp://0.0.0.0:4840/opcua/")
    uri = "http://mycompany.org/OPCUA"
    idx = await server.register_namespace(uri)
    await server.import_xml('customobject.xml')

    device_type = (await server.nodes.base_object_type.get_child([ f"{idx}:ExampleType" ]))
    object_node = await server.get_objects_node().add_object(idx, "This Is A Custom Object", device_type)
    uint16_object_node = await object_node.get_child([f"{idx}:ThisIsUInt16"])
    
    # This fails with --> "Write refused: Variant: Variant(Value=12345, VariantType=<VariantType.Int64: 8>, Dimensions=None, is_array=False) with type VariantType.Int64 does not have expected type: VariantType.UInt16"
    #await uint16_object_node.set_value(12345)

    # This fails with --> "Write refused: Variant: Variant(Value=Variant(val:12345,type:VariantType.UInt16), VariantType=<VariantType.Variant: 24>, Dimensions=None, is_array=False) with type VariantType.Variant does not have expected type: VariantType.UInt16"
    #await uint16_object_node.set_value(ua.Variant(12345, ua.VariantType.UInt16))

    # This fails with --> "Write refused: Variant: Variant(Value=DataValue(Value:Variant(val:12345,type:VariantType.UInt16), StatusCode:StatusCode(Good)), VariantType=<VariantType.DataValue: 23>, Dimensions=None, is_array=False) with type VariantType.DataValue does not have expected type: VariantType.UInt16"
    #await uint16_object_node.set_value(ua.DataValue(ua.Variant(12345, ua.VariantType.UInt16)))

    # This fails with --> "Write refused: Variant: Variant(val:12345,type:VariantType.UInt16) with type VariantType.UInt16 does not have expected type: VariantType.UInt16"
    #await uint16_object_node.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(12345, ua.VariantType.UInt16)))

    # Getting the value data type
    datavalue = await uint16_object_node.read_data_value()
    print(datavalue) # <-- it prints "DataValue(Value=Variant(Value=0, VariantType=<VariantType.UInt16: 5>, Dimensions=None, is_array=False), StatusCode_=StatusCode(value=0), SourceTimestamp=datetime.datetime(2021, 9, 14, 9, 22, 51, 346231), ServerTimestamp=None, SourcePicoseconds=None, ServerPicoseconds=None)""
    # This is OK
    await uint16_object_node.write_value(datavalue)

    # starting!
    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())
    
```

This is the **customobject.xml** file:
```
<?xml version='1.0' encoding='utf-8'?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <NamespaceUris>
    <Uri>http://mycompany.org/OPCUA</Uri>
  </NamespaceUris>
  <Aliases>
    <Alias Alias="UInt16">i=5</Alias>
    <Alias Alias="HasModellingRule">i=37</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasSubtype">i=45</Alias>
    <Alias Alias="HasProperty">i=46</Alias>
  </Aliases>
  <UAObjectType NodeId="ns=1;i=2001" BrowseName="1:ExampleType">
    <DisplayName>ExampleType</DisplayName>
    <Description>ExampleType</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=2010</Reference>
    </References>
  </UAObjectType>
  <UAVariable NodeId="ns=1;i=2010" BrowseName="1:ThisIsUInt16" ParentNodeId="ns=1;i=2001" DataType="UInt16">
    <DisplayName>ThisIsUInt16</DisplayName>
    <Description>ThisIsUInt16</Description>
    <References>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=2001</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
    </References>
    <Value>
      <uax:UInt16>0</uax:UInt16>
    </Value>
  </UAVariable>
</UANodeSet>
```

AndreasHeine, 2021-09-14:
typeguessing can not detect the difference between Int16 Int32 and Int64 !!!

```python
        # always specify the correct VariantType!
        dv = ua.DataValue(ua.Variant(200.0, ua.VariantType.Double))
        print("Write to the node:")
        print(dv)
        await double_node.write_value(dv)
```

https://github.com/AndreasHeine/opcua-tutorial/blob/main/client/write.py

xmas79, 2021-09-14:
@AndreasHeine thanks for your quick reply!

Unfortunately it doesn't work even specifying VariantType.UInt16. Could you please have a look at the code I already posted and maybe spot if I made something wrong?

Thanks!

AndreasHeine, 2021-09-14:
```python
import asyncio
# import opcua.ua as ua <-----------------------------! 

from asyncua import Server, ua # 
```

you are mixing python-opcua and asyncua ... ;) do not!

xmas79, 2021-09-14:
@AndreasHeine That was!!!

closed 2021-09-14 by xmas79