FreeOpcUa/opcua-asyncio #689: is_array is False even if the variant is an array

ybozer, 2021-10-17:
**Describe the bug** <br />
"is_array" of the Variant is set to "False" for arrays. 

**To Reproduce**<br />
1. Add an array variable in server.
2. Read the array by using server.read_attribute_value() method.
3. Check the "is_array" attribute of the array.

**Expected behavior**<br />
The array's "is_array" attribute is True.

**Screenshots**<br />
[image]

[image]

**Version**<br />
Python-Version: 3.7.3<br />
opcua-asyncio Version (e.g. master branch, 0.9): master branch

AndreasHeine, 2021-10-17:
https://github.com/FreeOpcUa/opcua-asyncio/blob/a69566c5ae926233e85be3b77c4f5480f148614e/asyncua/ua/uatypes.py#L783

looks like something is broken since Dataclass update...

AndreasHeine, 2021-10-17:
https://github.com/FreeOpcUa/opcua-asyncio/blob/a69566c5ae926233e85be3b77c4f5480f148614e/asyncua/ua/uatypes.py#L842

if i use value = [0,1,2,3,4,5]
this looks wrong -> dims = [6]

AndreasHeine, 2021-10-17:
https://github.com/FreeOpcUa/opcua-asyncio/blob/a69566c5ae926233e85be3b77c4f5480f148614e/asyncua/ua/uatypes.py#L804

should be "none" i guess

oroulet, 2021-10-18:
I tried to add back the is_array for compatibility but it adds a lot if complicated if sentences.

oroulet, 2021-11-11:
one more attempt to fix this correctly. Basiclaly I did what @AndreasHeine had in his PR apart from that dims > 1 stuff. As far as I understand we do not need to set that dimension variable unless requested by user or if we have more than one dimension. In fact setting that dimension seems to break uaexpert. Please test

oroulet, 2021-11-11:
https://github.com/FreeOpcUa/opcua-asyncio/pull/717

oroulet, 2021-11-11:
test master I merged since it was broken

closed 2023-03-29 by schroeder-